prusa3d/PrusaSlicer #7872: Keep Staggering of Seams as in 2.4.0

stefan-am, 2022-02-03:
### Version
PrusaSlicer 2.4.1-beta1

### Operating system type + version
Win 10 64 bit

### Behavior
I have been very happy to see PrusaSlicer 2.4.0 now doing this "Staggering" of seams, as it creates parts that are much stronger. 
Seams of parts that are mechanically stressed should never line up (see first picture, desired behaviour) 
Ideally the seams of the perimeters should interlock like the teeth of a zipper.

Now the documentation for 2.4.1 describes that this will not be done any more and behaviour is back to pre-2.4.0 (2. and 3. pic - undesired behaviour) 

This is not an improvement but a step back for everyone who wants to create strong functional parts.
I personally did not realize any downside of the 2.4.0-staggering-behaviour and parts and seams looked PERFECT and better than ever before!

_Is this a new feature request?_ 
Maybe... Either the behaviour should be kept as in 2.4.0 or there should be a option "seams staggering yes/no". Maybe there should be a parameter that controls the offset of the seam positions in the different perimeters. 

[image]

[image]

undesired behaviour:
[image]

bubnikv, 2022-02-03:
We found some issues with the seam staggering. We will re-evaluate seam staggering for 2.5.0.

Also we are working on seam placement in general for 2.5.0.

n8bot, 2022-02-03:
The seam placement tool could be more like a pencil or razor blade. The painting is not accurate enough for small nozzles or precise placement. I'm sure you will come up with something useful.

dimo414, 2023-06-26:
I notice in 2.6.0 there's a ["Staggered inner seams" option](https://help.prusa3d.com/article/layers-and-perimeters_1748#staggered-inner-seams) (link appears broken atm) - is this issue resolved now?

stefan-am, 2023-06-26:
i am happy that this option is available again!
as i said in my first post:

 _Maybe there should be a parameter that controls the offset of the seam positions in the different perimeters._
 
But for now i can work with the default quite fine.

kubispe1, 2023-08-27:
I am closing this issue. Staggered seams are available (not by default) but as optional settings. 
If I overlooked something in this issue let me know. 
Thanks and have a nice day

closed 2023-08-27 by kubispe1

dimo414, 2023-08-27:
@kubispe1 FWIW the "Staggered inner seams" option in PrusaSlicer 2.6.0 links to https://help.prusa3d.com/article/layers-and-perimeters_1748#staggered-inner-seams which is still a broken link.

psiberfunk, 2024-03-08:
> @kubispe1 FWIW the "Staggered inner seams" option in PrusaSlicer 2.6.0 links to https://help.prusa3d.com/article/layers-and-perimeters_1748#staggered-inner-seams which is still a broken link.

@kubispe1  the documentation is still missing...