prusa3d/PrusaSlicer #7873: Do not center models imported from prusaslicer 3MF files

nordurljosahvida, 2022-02-03:
### Version
2.4.0

### Operating system type + version

Ubuntu 20

### Behavior
* _Describe the problem_

when importing geometry only from another prusaslicer 3MF file, version 2.3.0 correctly imported it as is, without centering the geometry. 2.4.0 auto-centers the imported geometry without asking. since the model was previously sliced in that way, it makes no sense and is detrimental to move the object on the plate as they were likely placed that way for a reason

* _Steps needed to reproduce the problem_

from any open instance in 2.4.0 ctrl-i / import prusaslicer-generated 3MF where geomtries were not centered, they'll be auto-centered in the process

* _Expected Results_

no auto-centering. there are numerous cases where it is simpler to have an open instance, delete all files, and import geometry from an older prusaslicer-generated 3MF file rather than opening that file, ctrl-L to import config only from the newer file. don't break this flow please.

* _Actual Results_

geometry will be auto-centered

_Is this a new feature request?_

nope

nordurljosahvida, 2023-06-18:
could anyone please look into this? This is a regression from a previously working condition IMO. Thanks in advance!

bubnikv, 2023-06-19:
I suppose we don't follow your issue. Maybe giving examples by screenshots
and 3MFS would help to clarify your issue.

nordurljosahvida, 2023-06-19:
Thanks @bubnikv, basically when doing importing geometry only from a PS-generated 3MF, position was maintained until 2.3.0. From 2.4.0, the imported geometry is auto-centered, which is IMO incorrect as position should be kept from the imported file's geometries, as has always been.

For example, this is a daily occurrence for us:

Take this example where you import multiple STLs, some of which outside the build area. This happens extremely often when you have multiple objects to print and some don't fit. PS btw automatically arranges multiple plates side by side as in the picture below, which is a fantastic feature:

[image]

Here you have printed the first batch and are now preparing to print the second batch as in the picture above. Let's say, instead of simply opening the original project, you only want to import the geometry from the saved 3MF, for any number of reasons, for example you already have a different project open for batch 2 with new slicing settings and filaments.

You'd usually have simply imported geometry only from the original 3MF and bam, you're done. As you've always done, you go and hit ctrl-I and select the previous 3MF [or drag and drop and select geometry only import].

2.3.0 correctly parsed the file as a PS-generated 3MF and kept the position. Now instead, the model is autocentered:

[image]

This requires manual re-alignment and is generally annoying as the geometry was previously created inside prusaslicer and positioned exactly ready for printing. This is especially bothersome when the models barely fit in the plate and you have to spend 30 seconds simply visually aligning the parts with the buildplate.

So to sum up: when importing geometry from a 3MF, we shouldn't be auto-centering the imported models [if the origin 3MF was PS-generated, *at least*].

This ease of use was broken in 2.4.0.

Thanks again
[arrangement.zip](https://github.com/prusa3d/PrusaSlicer/files/11788433/arrangement.zip)

closed 2026-09-29 by prusaslicer-bot[bot]