prusa3d/PrusaSlicer #7877: change of depth of multimaterial painting

Sunny953, 2022-02-04:
### Version
_2.4.0_

### Behavior
I search for an option of the multimaterial painting, where I could tell the slicer how deep the painting should be.
It could be an advantage to save time, if there is as less color(changes) as possible.
For example: It would be great if I could say that the painted colorarea should only be 0,8mm deep or if I like to have an transparent area, that I could exceed the depth.

Is this possible?

_Is this a new feature request?_
YES

hejllukas, 2022-02-04:
Hi, there is an option in the slicer called **Maximum width of a segmented region** (Print Settings->Multiple Extruders->Advanced) that allows you to set the maximum depth globally for all painted regions. 
I have attached two examples below to illustrate how it works.

| Maximum width of a segmented region=0 (off) | Maximum width of a segmented region=5mm |
| --- | --- |
| [image] | [image] |

Sunny953, 2022-02-04:
Thanks for this fast support and sorry for my "stupid" question...didn't know that before!

closed 2022-02-04 by Sunny953