prusa3d/PrusaSlicer #7878: Using CLI to generate gcode files does not create embedded thumbnails

jwkicklighter, 2022-02-04:
### Version

Version 2.4.0+arm64

### Operating system type + version

- MacOS 11.2.3
- Mac Mini (M1, 2020)

### 3D printer brand / version + firmware version (if known)

Original Prusa i3 MK3S+ (firmware version not applicable to problem)

### Behavior

1. Modify the MK3S profile to include thumbnail support
2. Generate a gcode file
3. Inspect the gcode file, there will be a thumbnail section
4. Export the config that was just used
5. With the CLI, use the config that was just exported to slice the same file
6. Inspect the gcode file, there **will not** be a thumbnail section

#### Project File (.3MF) where problem occurs

This is repeatable across any 3MF/STL that I have tried.

neophyl, 2022-02-05:
Doesn’t it use the gui to create the embedded thumbnail ?  If it’s not running the gui then ….

jwkicklighter, 2022-02-06:
@neophyl honestly I have no idea how it works at the technical level. I didn't think that the UI was used because the thumbnail is always from the same perspective/zoom regardless of the UI view. But if that's the case, it seems like a headless renderer could be started up to generate the image data.

At a minimum, this should be documented as not being supported. I'm hoping that it can actually be enabled, though.

jwkicklighter, 2022-02-08:
After further investigating, the CLI help actually says that this option is supported when using `--help--fff`:

```
--thumbnails        Picture sizes to be stored into a .gcode and .sl1 / .sl1s files, in the
                    following format: "XxY, XxY, ..." (default: )
```

I have tried to supply the thumbnail config as an argument to the CLI rather than just in the config `.ini` file, but it also does not work. Perhaps notable, the comment at the bottom of the generated `.gcode` file *does* show the `thumbnails` option as I have it configured:

```
; thumbnails = 16x16,220x124
```

bubnikv, 2022-02-09:
The command line slicer does NOT support thumbnails. The thumbnails are
generated by the same OpenGL context that the 3D scene is rendered with.
There is no OpenGL context available as of now in command line slicer, thus
there are no thumbnails. We should remove the --thumbnails option from CLI,
it is generated automatically.

bubnikv, 2022-02-11:
> @bubnikv <https://github.com/bubnikv> Would there be any way to use an
> already-running GUI process to render the thumbnail when using the CLI? For
> my needs, I'm using the CLI for automation purposes (not having to manually
> slice hundreds of files) so a speed decrease would not matter. I think it
> would be reasonable that someone wanting to generate thumbnails would
> expect it to take longer to process each file.

It is doable, but it does not fit the current architecture. It would
require quite a lot of code and we have higher priority tasks.

jwkicklighter, 2022-02-11:
@bubnikv Understandable, thank you for the info

Torrax, 2022-03-08:
I also think this would be an awesome addition. Maybe to mark as a feature request to look at some point in the future if time permits?

TomasHubelbauer, 2022-07-12:
I'm looking for the same thing but not for GCode but for the 3MF. The GUI export does include `Metadata/thumbnail.png` in the 3MF, the CLI export doesn't. CLI command I used is in #8492.

jkavalik, 2022-07-12:
What about implementing some simple software rasterizer instead, so OpenGL is not needed at all? It could do useful low-res thumbnails too. I was checking some possibilities and found https://github.com/aslze/minirender as an example.

demonlibra, 2022-10-24:
I hope the author will continue to develop the functionality of the project [minirender](https://github.com/aslze/minirender)

jimb15122, 2022-11-04:
Suggestion: since the thumbnail is just a folder with a few png's in it, how about adding a CLI option to specify a LAYER to use as the thumbnail.  eg: --thumbnail-layer=125.   I could care less about the 3d perspective.  I just need to verify that I opened the right file before starting the print.  The doc for this option could also explain that normal thumbnails are not supported.

jimb15122, 2022-11-07:
UvTools version 3.8.2 provides a workaround for the missing thumbnails.  You can set the thumbnail to any specified layer or to the heatmap (avg pixel shade of all layers), or to a specified image file.  This is available from both the GUI and CLI versions. 

e.g. to set the thumbnail to layer 50, use:
UvToolsCmd.exe set-thumbnail my.sl1 50

See UvTools issue # 599 for details.

jkavalik, 2022-11-07:
That sounds nice for SLA printers but won't work for FDM gcode files afaik.

wolfallein, 2022-12-04:
I also think that a thumbnail in the CLI version is very useful, I hope it can be integrated in the future.

I didn't know it relies on a screenshot from the GUI.

nat-a-cyborg, 2023-03-23:
What would be awesome is if we could pass our own thumbnail image. I'd honestly do this instead of the automatic thumbnails. I'll post back here if I implement this.

Maxime-JULLIEN, 2024-12-11:
It could be cool to allow the cli to generate thumbnails. It's really important.

closed 2026-09-29 by prusaslicer-bot[bot]

Pearcekieser, 2026-10-01:
Could this issue be reopened for the remaining 3.0 CLI thumbnail case?

Current 3.0 code can reuse an embedded 3MF preview, but a valid project without `Metadata/thumbnail.png` still exports G-code without thumbnails even when thumbnail sizes are configured.

In testing on macOS 26.5.1 / arm64 against the 3.0 development code based on `30ef5919`, the missing-preview case also failed with the prerequisite renderer port alone. The fallback in [PR #15909](https://github.com/prusa3d/PrusaSlicer/pull/15909) resolves it; reproduction coverage and validation details are documented there.

I understand that earlier comments described CLI rendering as unsupported. Would you consider this remaining case a bug appropriate for reopening here, or should it be tracked as a feature request in Discussions?

Update: I opened [#15937](https://github.com/prusa3d/PrusaSlicer/issues/15937) to track the remaining 3.0 missing-preview case, with the validation limits and a link back to this report. My account cannot reopen this issue.