prusa3d/PrusaSlicer #7881: Variable Layer Height feature not available on AzteQ Industrial or DeltiQ 2 printers

parhelia, 2022-02-05:
### Version
Version 2.4.1-beta1+x64

### Operating system type + version
mac OS Monterey 12.2 (21D49)

### 3D printer brand / version + firmware version (if known)
DeltiQ 2 full stock

### Behavior
Variable layer height feature is available for all Prusa FDM printers, Creality printers. Feature is grayed out and not selectable on DeltiQ 2 or AzteQ Industrial.

Step 1. Choose either "DeltiQ 2" or "AzteQ Industrial" from the drop-down menu.
Step 2: Import a 3MF or an STL model

_ Expected Result _
Variable layer height icon should be enabled and can be selected
[image]

_ Actual Result _
Variable layer height icon is disabled and grayed out, and cannot be selected.
[image]

Jebtrix, 2022-02-05:
Is Printer Settings > General > Advanced > Enable variable height feature turned on?

parhelia, 2022-02-05:
That was the solution. Thank you. I did not realise it was a printer-specific setting. I will close this issue.

closed 2022-02-05 by parhelia