prusa3d/PrusaSlicer #7882: Can't access /dev/ttyACM0 in Ubuntu 20.04

sumnermichaelsmith, 2022-02-05:
### Version
2.4.0

### Operating system type + version
Ubuntu 20.04
<Prusa-slicer installed via snap>

### 3D printer brand / version + firmware version (if known)
Original i3 MK3S+
3.10.0-4481

### Behavior
I cannot access the printer from prusa-slicer to upgrade firmware.  The log message is:

>>>>>>
avrdude-slic3r -v -p atmega2560 -c wiring -P /dev/ttyACM0 -b 115200 -D -U flash:w:0:/home/smichaelsmith/Downloads/prusa3d_fw_MK3S_MK3S+_3_10_1_4697.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Jan 30 2022 at 15:49:14
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : /dev/ttyACM0
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
avrdude-slic3r: ser_open(): can't open device "/dev/ttyACM0": Operation not permitted
avrdude-slic3r: Could not open port: /dev/ttyACM0

avrdude-slic3r done.  Thank you.
<<<<<<

I have user permission to access "dialout", which is where ttyACM0 connects, but the error message above appears to indicate insufficient privilege.  Due to the installation, prusa-slicer is located in /snap/bin/prusa-slicer, and otherwise works as expected.  Could this be a user change related to 'avrdude-slic3r'?  I must presume that 'avrdude-slic3r' is located within 'prusa-slicer', as it does not appear elsewhere, but does appear to run.  Does it change user or permissions?  If so, how can I know the user to add to groups?

I have previously done an update using the same computer hardware, but with an older version of the O/S, so I know the hardware works.  'lsusb' gives:
Bus 002 Device 002: ID 8087:8001 Intel Corp. 
   ***
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
>>>Bus 003 Device 012: ID 2c99:0002 Prusa Research (prusa3d.com) Original Prusa i3 MK3<<<
Bus 003 Device 007: ID 05e3:0608 Genesys Logic, Inc. Hub
   ***

'ls -l /dev/ttyACM0' shows:

` crw-rw---- 1 root dialout 166, 0 Feb  5 12:58 /dev/ttyACM0`

'group' shows:
`dialout:x:20:smichaelsmith`  (I am user 'smichaelsmith')`

#### Project File (.3MF) where problem occurs
does not apply

Jebtrix, 2022-02-06:
Have you tried `sudo chmod a+rw /dev/ttyACM0`? You could also try (re)installing snap with `sudo snap install prusa-slicer --devmode`

sumnermichaelsmith, 2022-02-07:
Thanks.  The re-installation of snap module 'prusa-slicer' appears to 
have fixed the problem.  I first tried changing permissions on ttyACM0, 
but that didn't help.  Then I removed and installed prusa-slicer in 
--devmode, and the result was successful (although I did still have 
ttyACM0 permissions set for 'all' rw; I don't think that was necessary).

=m

closed 2022-02-07 by sumnermichaelsmith

Jebtrix, 2022-02-07:
I took the liberty to forward this issue to the snap author.

https://github.com/ivocavalcante/PrusaSlicer-snap/issues/7

ivocavalcante, 2022-02-07:
Hi people,

@Jebtrix thanks for bringing this to me.

@sumnermichaelsmith I believe you're having problems with Snap sandboxing. Installing in devmode isn't really a solution, since it effectively bypass all security features.

Could you, please, try installing in normal mode again and issue following command:

`snap connect  prusa-slicer:raw-usb`   (you might need to use `sudo`, if not logged in to the store)

This interface is needed so the Snap can access USB ports. I had this described in the store listing but, looking at it now, it seems to have slipped on some update. I'd appreciate if you could test and give feedback, so I can include it in the docs again.

Thanks,
Ivo

sumnermichaelsmith, 2022-02-07:
I removed, installed prusa-slicer normally, ran the command you suggested, ran 'snap restore...', then ran prusa-slicer and re-flashed the printer.  This seems to work fine.  The re-flash was successful.  Thanks!

ivocavalcante, 2022-02-07:
Great! Will update the doc on store listing. Thanks!

JesseObrien, 2023-01-14:
This worked for me on Ubuntu 22.04.01 LTS. Thanks!

`snap connect  prusa-slicer:raw-usb`

patate1684, 2023-01-28:
This worked for me too on Ubuntu 22.04.1 LTS. Thanks!
`snap connect  prusa-slicer:raw-usb`

tills13, 2024-10-17:
FYI in addition to the above I had to run 

```
sudo chmod a+rw /dev/ttyACM0
```

And now it's working 🎉 thanks everyone.