prusa3d/PrusaSlicer #7885: Broken slicing of cylinder 100% infill

LightningShark25, 2022-02-06:
### Version
2.4.0

### Operating system type + version
Win10

### Behavior
This model has been sliced really oddly (badly).   It seems to show internal infill, solid infill and top infill, all broken up in differnt stages of the print on the same layer.  The object is set to 100% infill.
The object is a pipe elbow.
Refer to the screenshot.

[image]

Jebtrix, 2022-02-06:
It would be better if you zipped and attached this project so people may review it properly. I'm assuming you want the slice to look more like this inside the perimeters: 

[image]

[image]

LightningShark25, 2022-02-06:
I'll do that later tonight..

It was late when I posted that.. I can sorta understand why the slicer has broken up top and internal infill.. 

That said it doesn't make structural sense to slice it that way.. One workaround was to do 100 perimeters but this does not print well.  Or will it do well for integrity as there is no cross pattern 

Not necessarily calling it a bug but there is some optimising that could be done.. But I'll admit I have no idea how.

I have structural reasons for printing in this orientation BTW. As the stress is pushing the ends of the pipe apart

Jebtrix, 2022-02-06:
For the record when I do 100% infills, the settings in previous post is always used. Using _sparse infills_, enforcing horizontal shell thickness can help with its intended function. But at 100% infill I'd go as far to say these settings should be auto recommended/set.

Jebtrix, 2022-02-06:
Just a side note you can't set bottom pattern doing it with my settings, so its not the be all end all workaround.

bubnikv, 2022-02-07:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/107

LightningShark25, 2022-02-07:
Thanks @bubnikv this duplicate seems to align with what is going on... Although I don't recall specifying different width extrusion for solid vs top infills etc.  I will check if I can tweak it somehow

PaparazziN, 2023-10-22:
Hi! @LightningShark25!
Did you check if it made any difference adjusting extrusion width? Regardless, if you agree that it is a duplicate of #107, it might be worth closing this issue.

closed 2026-09-29 by prusaslicer-bot[bot]