prusa3d/PrusaSlicer #7888: Infill of walls issue

KFabricius, 2022-02-07:
### Version
Version 2.4.0+win64

### Operating system type + version
PrusaSlicer
Version:   2.4.0+win64
Build:     PrusaSlicer-2.4.0+win64-202112211614

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 22000), 64-bit edition
Total RAM size [MB]: 33,506MB
OpenGL installation
GL version:   4.6.0 NVIDIA 472.88
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA T1200 Laptop GPU/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 341 lines cut ...]
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

### 3D printer brand / version + firmware version (if known)
Snapmaker for now...

### Behavior
Issues with slicing:
There are strange slicing going on with the infill in the walls at layer 790, see included 3MF file.

#### Project File (.3MF) where problem occurs
Please see included files:

(https://user-images.githubusercontent.com/99171771/152752267-c38a1eee-209f-4af6-9d6d-c9e708e6e2d9.png)
[Case.zip](https://github.com/prusa3d/PrusaSlicer/files/8013553/Case.zip)

KFabricius, 2022-02-07:
[image]

Jebtrix, 2022-02-07:
Thin wall detection problem like #4412 and many others. For your particular object just turn it off.

KFabricius, 2022-02-08:
Hi Jextrix, thx I do realize that you can switch off thin wall detection, it only took me 1 hour to figure it out, that said:

1. New users like me do not always understand what causes issues like this, in fact, I more or less expect to press slice and get a good slice out of the model. I do actually work within the CAD/CAM industry, and I am printing as a hobby.

2. As I wrote then I do work in the CAD/CAM industry, and we would for sure look at an issue like this one if it was our software, as it will cause many support issues and worse of all bad prints.

Regards,
Kris

KFabricius, 2022-02-09:
> @KFabricius
> 
> > As I wrote then I do work in the CAD/CAM industry, and we would for sure look at an issue like this one if it was our software, as it will cause many support issues and worse of all bad prints.
> 
> We took your advice and we are looking into this issue diligently. Indeed, we have one full time developer dedicated to it.

- thx for the feedback, much appreciated.
- it was not means to be a pain in the ass as I know you have limited resources but more a heads up as this is a very important issue to fix in my opinion. Anyway, keep up the good work and thx again for listening!
- BR Kris

closed 2026-09-29 by prusaslicer-bot[bot]