prusa3d/PrusaSlicer #7889: Purge block code contains unneccesary double/triple spaces before values

DeeKay64, 2022-02-07:
### Version
Prusaslicer up until 2.4.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Lotmaxx Shark

### Behavior
Prusaslicer adds unnecessary double and triple spaces in front of values in the purge block code, which it shouldn't. Ofcourse this also should not make a difference for the printer if it adhered to the standard, but alas, the Lotmaxx Shark with its heavily hacked Marlin Frankenstein firmware just doesn't for whatever reason, so it ignores all G1 commands with multiple spaces and no purge blocks are printed. Even though the FW source has been released nobody was able to compile the firmware because of the highly proprietary toolchain that Lotmaxx used, so the users that want to use Prusaslicer always need to postprocess the Gcode and remove double/triple spaces in order for Gcodes to work on this printer. Hence it would be great if Prusa solved this issue on their side, so people aren't stuck with Cura!

Actual results example:

; CP TOOLCHANGE WIPE
G1  X243.344 Y198.383  
G1  X184.344  E2.5285 F1600
G1  X185.094 Y197.883   E0.0386
G1  X243.844  E2.5178 F1800
G1  X243.094 Y197.383   E0.0386
G1  X185.094  E2.4856 F2200
G1  Y196.883  E0.0214
G1  X243.094  E2.4856 F4200
G1  Y196.383  E0.0214
G1  X184.344  E2.5178 F4250
G1  X185.094 Y195.883   E0.0386
G1  X243.844  E2.5178 F4300
G1  X243.094 Y195.383   E0.0386
G1  X185.094  E2.4856 F4350
G1  Y194.883  E0.0214
G1  X243.094  E2.4856 F4400
G1  Y194.383  E0.0214
G1  X184.344  E2.5178 F4450
G1  X185.094 Y193.883   E0.0386
G1  X243.844  E2.5178 F4500
G1  X243.094 Y193.383   E0.0386
G1  X185.094  E2.4856 F4550
G1  Y192.883  E0.0214
G1  X243.094  E2.4856 F4600
G1  Y192.383  E0.0214
G1  X184.344  E2.5178 F4650
G1  X185.094 Y191.883   E0.0386
G1  X243.844  E2.5178 F4700
G1  X244.094 
G1  Y200.443 
G1  X184.094  E2.5713
G1  Y162.763  E1.6148
G1  X244.094  E2.5713
G1  Y200.443  E1.6148
M220 R
G1 F6000
G4 S0
G92 E0
; CP TOOLCHANGE END
;------------------

Expected results example:

; CP TOOLCHANGE WIPE
G1 X243.344 Y198.383 
G1 X184.344 E2.5285 F1600
G1 X185.094 Y197.883 E0.0386
G1 X243.844 E2.5178 F1800
G1 X243.094 Y197.383 E0.0386
G1 X185.094 E2.4856 F2200
G1 Y196.883 E0.0214
G1 X243.094 E2.4856 F4200
G1 Y196.383 E0.0214
G1 X184.344 E2.5178 F4250
G1 X185.094 Y195.883 E0.0386
G1 X243.844 E2.5178 F4300
G1 X243.094 Y195.383 E0.0386
G1 X185.094 E2.4856 F4350
G1 Y194.883 E0.0214
G1 X243.094 E2.4856 F4400
G1 Y194.383 E0.0214
G1 X184.344 E2.5178 F4450
G1 X185.094 Y193.883 E0.0386
G1 X243.844 E2.5178 F4500
G1 X243.094 Y193.383 E0.0386
G1 X185.094 E2.4856 F4550
G1 Y192.883 E0.0214
G1 X243.094 E2.4856 F4600
G1 Y192.383 E0.0214
G1 X184.344 E2.5178 F4650
G1 X185.094 Y191.883 E0.0386
G1 X243.844 E2.5178 F4700
G1 X244.094 
G1 Y200.443 
G1 X184.094 E2.5713
G1 Y162.763 E1.6148
G1 X244.094 E2.5713
G1 Y200.443 E1.6148
M220 R
G1 F6000
G4 S0
G92 E0
; CP TOOLCHANGE END
;------------------

bubnikv, 2022-02-07:
I think this is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7503

DeeKay64, 2022-02-07:
Yeah, that's a duplicate, can be merged/closed, thanks! ;-)

closed 2022-02-07 by bubnikv