prusa3d/PrusaSlicer #7891: Different kind of heat creep

fungreenfox, 2022-02-07:
### Version
_Version of PrusaSlicer used goes here_ 
Version 233

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Ender 6 with firmware from Th3dstudios.com version 256K firmware
Mainboard creality 4.3.1

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

With BLtouch firmware 149, the best slicer was this prusa. Other slicers would mess up the print in weird ways. But prusa just did its job.
_Is this a new feature request?_
With the new firmware from Th3dstudios, the issue is explained as "head creep", where the slicer stops feeding filament after ca ½ cm of z-axis layer shift. (how tall is your printing object)
But ultimaker cura can make the print job done, so whats this difference, that prusa is failing now?
Is there an option for layer shift and filament?

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
Is not possible to upload here. The file is linked to dropbox.
https://www.dropbox.com/s/1rt8c5m0tts4nka/ezabl%20holder.zip?dl=0

lukasmatena, 2022-02-08:
I don't know. The retraction length of 30 mm looks quite suspicious. It looks like a configuration problem to me. You can try to upgrade to PrusaSlicer 2.4.0 and reslice using the Ender-6 profile provided by us (top menu -> Configuration -> Configuration Wizard -> Other Vendors), or at least inherit your profile from that before you tweak it. But I don't know whether it will work with "firmware from Th3dstudios.com version 256K firmware", whatever that might be.

fungreenfox, 2022-02-19:
Have not been using the ender 6 profile, but just doing it custom. If there is some differense in using that, it should be tried.

(The firmware is custom for ender 6, when getting their lcd display and probe (that doesn't touch the hotbed).
This adds a prusa like display with more options, and a fast probe).

rtyr, 2022-03-10:
Can we close this issue?

rtyr, 2022-03-30:
No response. Likely misconfigured retraction settings. Closing.