prusa3d/PrusaSlicer #7892: Add possibility to add more custom G-Code

U-B-Glade, 2022-02-07:
### Version
V2.4.0+win64

### Operating system type + version
W10 20H2

### 3D printer brand / version + firmware version (if known)
MK3s MMU2 3.10.1 1.0.6
### Behavior

I like to insert more custom G-Code e.g. at:
; CP TOOLCHANGE UNLOAD -> e.g. M109 S{temperature[0]-10}
; CP TOOLCHANGE LOAD
; CP TOOLCHANGE END

_Is this a new feature request?_
Yes 

It would offer the possibility to use MMU with Multi Material instead of Multi-Color.
And it would solve the issue, that filament tips are sometimes to hot (Stringing), in particular if I like to print with the max temp for strongness.

fe60, 2022-02-08:
PrusaSlicer 2.4.1 will offer [G-code substitutions](https://help.prusa3d.com/en/article/g-code-substitutions_301694) (already available in [2.4.1-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.1-beta1)).
Does this help?

U-B-Glade, 2022-02-08:
Hi fe60,

thanks for the info. 
I've tried it, but I get not familiar with it. The new substitution like to get Perl regular expressions like "${1}M221 S98n${2}M221 S95n${3}"
That is not wat I'm looking for. 
I mean real Custom G-Code like the existing in Printer Settings.
There are eight Fields now, so it is IMHO simple to generate three more. I'm wrong?

fe60, 2022-02-09:
**Feature request:**
I don't know how the development team decides to implement new features. But I guess it's not only about the efforts required to do so. I think the relevance for the (majority) of the users and usability do matter too. For rather special use cases the substitution feature has been added. Some users (incl. me) do such replacements already using text editors or ([post-processing](https://help.prusa3d.com/en/article/post-processing-scripts_283913)) scripts.

<details>
<summary><b>Using substitutions instead:</b></summary>

:warning: I don't own a MMU. I checked the related gcode for the first time and did not try out my proposal.
Your idea to pause the print and wait for the nozzle to cool down sounds problematic to me. At `; CP TOOLCHANGE UNLOAD` the nozzle seems to be close to the wipe tower. Instead of stopping you could slow down the wiping in the wipe tower. As the wiping of the "old" filament is pretty short, the example below slows down the wiping the new "new" filament.

Activate `Regular expression` for the substitution
Find: `(; CP TOOLCHANGE WIPE)`
Replace with: `\1\n; slow down\nM220 S60` (slow down to 60% speed)

There is no need to switch back to the default speed, as this is done by `M220 B` and `M220 R` commands already present in the generated gcode.
```
...
; CP TOOLCHANGE START
; toolchange #2
; material : PLA -> PETG
;--------------------
M220 B
M220 S100
; CP TOOLCHANGE UNLOAD
...
; select tool
T2
...
; CP TOOLCHANGE WIPE
; slow down
M220 S60
...
M220 R
G1 F10800
G4 S0
G92 E0
; CP TOOLCHANGE END
...
```

https://marlinfw.org/docs/gcode/M220.html
</details>

bubnikv, 2022-02-09:
> But I guess it's not only about the efforts required to do so. I think the relevance to (the majority of) the users and usability do matter too. 

Usability, maintainability and testing effort.

It would be lovely to have a user curated repository of post-processing scripts and regular expression replacements fitting various tasks.

traderhut, 2022-04-06:
(FYI: I updated the 8113 request, to show why it won't be solved by this request)

closed 2026-09-29 by prusaslicer-bot[bot]