prusa3d/PrusaSlicer #7893: [NRAI] Current Build, beta4, not reflected on splash.

foreachthing, 2022-02-08:
### Version
version_2.4.0-beta4-675-g199dc121a

Not really an issue, but: build beta4 does not show up in splash. It's still at beta1.

[image]

### Operating system type + version
Win10

plampix, 2022-02-08:
At 2.4.1-beta1. 2.4.0 got released Dec 22.

lukasmatena, 2022-02-08:
git describe returns beta4+675 after the merges we did around releases. We just tagged master as 2.5.0-alpha0 and switched configuration folder to PrusaSlicer-alpha again.

closed 2022-02-08 by lukasmatena

foreachthing, 2022-02-08:
Nevermind ... with a new pull/rebuild, it clearly shows 2.5.0-alpha0 in the title, splash and `git describe`.
Sorry to bother y'all!