prusa3d/PrusaSlicer #7898: 2.4.0: Only outer perimeters supported

jmding8, 2022-02-09:
Support interfaces are only being generated under the outer perimeter. They are not being generated under the inner perimeter, nor the solid infill

Bottom view of test part, support interfaces under the outer perimeter only:
[image]

For context, this is the part:
[image]

[image]

project file attached
[Shape-Cylinder.zip](https://github.com/prusa3d/PrusaSlicer/files/8036864/Shape-Cylinder.zip)

rtyr, 2022-02-10:
I don't see any issue. You have "thick bridges" enabled, and "detect bridging primeters" disabled so it is expected behavior.

[image]

The best result will be likely with "thick bridges" option disabled (or just use default profile).

jmding8, 2022-02-10:
Unchecking `Don't support bridges` still leaves the inner perimeter / bottom infill unsupported. Only enabling `detect bridging perimeters` applies the expected supports

n8bot, 2022-02-10:
I did notice a weird thing here though. Even with `thick_bridges` disabled, the bridge infill uses the "Bridge flow ratio" to set its width. Is this intended?

As in, if the ratio is 0.45 (as in the supplied 3mf from this issue) then the bridge infill is generated with a very tiny width. It doesn't seem intended.

jmding8, 2022-02-10:
> It explains the difference between perimeters and inner part. 

Does it make sense for the inner part to be not supported? The solid infill makes 180 degree U-turns where it joins the inner perimeter, so if it is not supported at the turn, it won't work properly.

> The supports under the perimeter are not generated closer to the middle because of your XY separation distance setting.

I see, you are right. Decreasing `XY separation distance` to 0.01 mm results in the inner perimeter being supported. However, it also puts the support right next to the wall of the object where it is likely to fuse. It doesn't make much sense to me that to keep the red zone clear, we need to increase `XY separation distance` to a higher number like >0.5mm, but to try to support the pink zone we need to decrease `XY separation distance` to a negative number like <-0.5mm.

[image]

`XY separation distance` should apply to the support relative to the stl surface. It should not apply to the boundaries of the lines / features being supported.

> And that is why we implemented "Thick bridges" option. If you disable it like I suggested, "standard" flow will be used instead of thick bridge flow used by default in earlier PS versions. It produces better surface finish on the bottom part.

I don't know if it's possible I have a bug in my version of 2.4.0, but if I check `Thick bridges`, no combination of `Don't support bridges` and/or `Detect bridging perimeters` results in the bottom solid infill being supported.

n8bot, 2022-02-10:
If you want the supports to be exactly like the 3mf you supplied, but with coverage under the solid infill, do as I said: load the 3mf you uploaded here, click no if asked about detect bridging perimeters, uncheck the "Don't support bridges" option, and slice, You will have support under the whole object.

If you want to slightly expand the supports underneath the inner perimeter, while maintaining the X/Y separation for the wall, use [my customized version of PrusaSlicer](https://github.com/n8bot/PrusaSlicer/releases/tag/version_2.4.0-release1-n8). It has an added setting called "Expand or filter support structures." The default value of -100% provides the standard PS behaviour. Increasing it will expand the supported areas, while respecting the X/Y separation.

n8bot, 2022-02-10:
Actually, this is weird. I'm getting inconsistent results when slicing this. Some times it slices perfectly without any changes to your 3mf... this is confusing.

n8bot, 2022-02-10:
Now, when I download your .zip, rename it 3mf, and click slice, I get this: 

[image]

And the Prusa MK3 bed model does not load correctly... there might be a problem with the profiles/3mf system in PS.

jmding8, 2022-02-12:
> If you want to slightly expand the supports underneath the inner perimeter, while maintaining the X/Y separation for the wall, use [my customized version of [...]

Have you tried to merge this into PrusaSlicer main? IMO, this is how supports should behave.

n8bot, 2022-02-12:
Yes, indeed I have. See PR: https://github.com/prusa3d/PrusaSlicer/pull/7131

lukasmatena, 2022-02-15:
To be honest, I am completely lost and don't understand what this issue is about, especially after @n8bot flooded it with reports of an issue that is not an issue and advertised his own fork along the way. It looks that there is several things mixed together.

Anyway, I sliced and inspected the project and see no problem. The inner perimeters are not supported because of **XY separation**. If you activate **Detect bridging perimeters**, they will be supported exactly as the bridging infill inside: that is, with **thick bridges** enabled exactly as in previous versions (it is a legacy parameter), with **thick bridges** disabled it will be spaced according to **top contact z distance**. From what I tested (and I admit I may have missed something), that is how it behaves. Which part of it is the problem?

(I tested in 2.4.1-beta1.)

jmding8, 2022-02-16:
First, I'd like to apologize for the tone of this issue. I was frustrated and my language was too harsh. Sorry for that.

Let me try to summarize the primary issue at hand. There is another issue that is related, but separate, and I will open a separate ticket for that. For now, let's focus on the below

## Thick bridges should not affect support generation
* To the user, it is not intuitive why **thick bridges** should interact with support settings. Bridging and support are two separate concepts: my bridging settings should not affect my support settings and vice versa.
* Furthermore, right now **thick bridges disabled** is de facto the only valid setting. Enabling **thick bridges** causes the support generation to fail in the following ways:
  * (1) inner perimeters and bottom infill not supported, if **detect bridging perimeters disabled**

[image]

*  * (2) nothing supported altogether, if **detect bridging perimeters enabled**

[image]

* These are not valid gcode and the print will fail. In 2022, PrusaSlicer should protect the user from producing invalid gcode like this.
* I understand the desire for backwards compatibility, but I dont see any reason why thick bridge enabled should cause support generation to produce these invalid results.

rtyr, 2022-02-16:
> Bridging and support are two separate concepts: my bridging settings should not affect my support settings and vice versa.

On the contrary, The "thick bridges" option was introduced mainly for better surface finish of the bottom most parts of the supported objects. Bridging is (and always was) directly related to printing with the supports. See https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1.

Bottom layers bridges

PrusaSlicer is based on the pioneering work of Alessandro Ranelucci and the RepRap community on the Slic3r project. As already mentioned above, the project was started ten years ago and a lot has changed since then indeed. The 3D printers are no more a rarity, ABS is not the only filament, motion systems are accurate and extruders are no more leaky. Thus adjustments to the slicing algorithms are called for.

Slic3r's strategies of FDM supports and object over support differ from the other FDM slicers. Namely, Slic3r was designed to make use of FDM bridging capabilities to circumvent the necessity of supports. And printing bridges with ABS was not that easy: The bridges were reliable only if extruding thick strands. Combining the thick bridges with supports required the 1st layer objects being printed thicker than the other layers, leading to ugly object undersides.

All the other FDM slicers evolved differently: The first layer over supports is printed with the same flow as the other layers, thus the first layer over supports looks nicer as long as the strands bridged reliably or supported by a support interface. For support materials that do not bridge reliably, dense support is required for high quality of objects over supports.

Starting with this release, PrusaSlicer newly offers printing bridges and 1st object layers over supports with the default layer thickness. The bridging behavior is configurable with a new "thick_bridges" boolean option. If enabled, the old Slic3r bridging strategy is applied, while if disabled, bridging is performed in a similar fashion to the other FDM slicers, leading to better bottom quality over supports. With "thick_bridges" disabled, the behavior of "bridge_flow_ratio" changes: Instead of modifying cross section of a rounded extrusion while keeping the gap between the extrusion threads constant, extrusion spacing is maintained while modifying the extrusion width and / or height to reach the prescribed extrusion ratio.

> Furthermore, right now thick bridges disabled is de facto the only valid setting. Enabling thick bridges causes the support generation to fail in the following ways

You have probably "don't support bridges" option enabled. This is how your project looks with thick bridges disabled, like I sugested in my first response.

[image]

This whole issue seems like a long discussion about the problem which doesn't exist.

jmding8, 2022-02-16:
Here is a video showing what I'm doing. Thick bridges enabled, detect bridging perimeters disabled. Both values of "dont support bridges" produce the same final support layer, where the inner perimeter is not supported.

https://user-images.githubusercontent.com/44815547/154313975-9cc33106-c7a8-456e-99d5-11274c2130de.mp4

Happy to make more videos capturing the behavior if you'd like. Seems like we are getting different behavior?

rtyr, 2022-02-16:
The video doesn't reflect what did you commented here https://github.com/prusa3d/PrusaSlicer/issues/7898#issuecomment-1041118380

[image]

Here you go, I just opened your project and disabled thick bridges:
[Shape-Cylinder2.zip](https://github.com/prusa3d/PrusaSlicer/files/8082287/Shape-Cylinder2.zip)

jmding8, 2022-02-16:
The problem is that thick bridges disabled is **de facto** the only valid setting. Either:
1. support generation should work correctly with both thick bridges **disabled** AND **enabled** 
2. or the user should not be able to toggle between thick bridges disabled / enabled and the "correct" choice should be the ONLY available option
3. or some UI element should be incorporated strongly encouraging the user to use thick bridges disabled

In 2022, for Prusa, making sure that the user cannot shoot themselves in the foot should be important. After all, doesn't Prusa brand itself as providing a more end-to-end user-friendly experience than Creality etc?

rtyr, 2022-02-16:
I deleted my previous comment as I misunderstood your previously. But I believe it was already explained that the reason for that behavior is XY separation distance setting.

rtyr, 2022-02-16:
I suggest to disable thick bridges and enable interface for the best result with this object.

TomaszSzt, 2022-04-11:
I must say I also got confused with those options especially when I used support enforcers. It was a relatively small part and it was surprising to me to see that it got supported near to the right place and then support was stopped and replaced with bridge hanging completely not supported at all. 

[image]
("don't support bridges" enabled, "thick bridges" enabled, "detect bridging perimeters" enabled). 

The result was that part was thicker than expected in this place even tough I believed it was supported. Which it was not.

[image] ("don't support bridges" enabled, "thick bridges" enabled, "detect bridging perimeters" disabled). This is the last support layer. A gap for "bridge" to hang down is clearly visible.

[image] ("don't support bridges" enabled, "thick bridges" disabled, "detect bridging perimeters" disabled).
This is the last support layer. There is no gap for a "bridge" so part seems to be correctly supported now.

I agree there may be a good reason for doing it like that, but it is really hard to guess, that "thick bridges"/"detect bridging perimeters" do have something with removing last support layer.

This is not an issue for me, because on big parts it is seen at once, but surely a slightly more informative tooltip would be expected.

closed 2026-09-29 by prusaslicer-bot[bot]