prusa3d/PrusaSlicer #7900: No Filamentdata in GCODE in octoprint

Devilmorpheus, 2022-02-10:
### Version 
1.4.1

### Operating system type + version
Win 10 & MAC IOs

### 3D printer brand / version + firmware version (if known)
Prusa MK3 & Mini

### Behavior
When I slice a STL in Prusa Slicer, I have the amount of Filament and the weight on the screen. After upload to Octoprint it‘s gone. It always says „0“

when I‘ll take The Cura Slicer it’s still there after the upload.

Does anybody know the reason for this Problem?

rtyr, 2022-02-10:
>  on the screen
Do you mean LCD on the printer?

The issue is likely not related to PrusaSlicer.

rtyr, 2022-02-10:
That doesn't look like standard MINI display :). 

It is likely caused by some Octoprint plugin. You can compare gcode generated by PrusaSlicer with the gcode file downloaded from Octoprint.

Devilmorpheus, 2022-02-10:
No it isn‘t, it‘s octodash :)

ok I will try that and see if there difference between both gcodes

closed 2026-09-29 by prusaslicer-bot[bot]