prusa3d/PrusaSlicer #7901: Slice is not generating the file as expected

DouglasOnisto, 2022-02-10:
### Version
2.4.0+

### Operating system type + version
Win10 64bits

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro (Latest stable marlin version)

### Behavior
I am trying to print a RC airplane, it needs to have small walls which is working as expected however there are some parts of the STL file that is not even showing on the preview version when I slice it. I already have the detect thin walls selected but it seems that other things are happening which is causing it to do not generated in the preview. I tried it thru Cura and it is working as expected but to be honest, I dont like Cura. If you compare the before and after screen attached, you will see that a lot of things are missing in the final model that is showing on the non sliced version.

_Is this a new feature request?_
Good question ><

#### Project File (.3MF) where problem occurs
[image]
[image]

I tried to upload the m3f file but it is not accepting here.

rtyr, 2022-02-10:
Please provide your 3MF project file, you have to archive it to the zip first.

DouglasOnisto, 2022-02-10:
[cessna_152_fuselage_5.zip](https://github.com/prusa3d/PrusaSlicer/files/8044571/cessna_152_fuselage_5.zip)

Added. Thanks

Jebtrix, 2022-02-11:
Couple setting changes to keep it as true as possible.
[expected_cessna_152_fuselage_5.zip](https://github.com/prusa3d/PrusaSlicer/files/8045320/expected_cessna_152_fuselage_5.zip)

DouglasOnisto, 2022-02-11:
Thank you @Jebtrix, I wasnt aware about this slice mode at all. I think I am good now =)

closed 2022-02-11 by rtyr