prusa3d/PrusaSlicer #7902: Prusa slicer installed on a surface pro x will not run properly 100%

Ender0331, 2022-02-10:
The install was clean and I can load the program just fine and operate all the menus and even set up my printer profiles and import bed skins, but...... when I load STL files it dose not show on the bed, anything to do with the bed dosnt seem to work, I can't rotate, I can't zoom in, I can't see the print nothing[image][image][image]

Jebtrix, 2022-02-11:
I'm don't think anything changed in Win11(yet) but https://docs.microsoft.com/en-us/surface/surface-pro-arm-app-performance states anything over OpenGL 1.1 is problematic. Recent article mentions lack of OpenGL support on arm :https://www.zdnet.com/article/op-ed-windows-isnt-ready-for-arm-developers/ . Sprinkle in x86 emulation under ARM... not surprising.

Have you tried [OpenCL™ and OpenGL® Compatibility Pack](https://www.microsoft.com/en-us/p/opencl-and-opengl-compatibility-pack/9nqpsl29bfff)?

Your best bet is probably just trying `--sw-renderer` cmdline switch launching Prusaslicer.

Ender0331, 2022-02-11:
Yes I'm already running opencl and it's still not running correctly

bubnikv, 2022-02-11:
Unfortunately we never ran PrusaSlicer on Windows on ARM. I also read https://www.zdnet.com/article/op-ed-windows-isnt-ready-for-arm-developers/  where it is stated that OpenGL is not supported at all, only DirectX. Thus I am surprised that your screenshots shows OpenGL by MESA, Microsoft being a vendor. I read there is a MESA OpenGL over Vulcan emulation, maybe Microsoft is now pushing that as an OpenGL solution.

closed 2026-09-29 by prusaslicer-bot[bot]