prusa3d/PrusaSlicer #7904: Support height and mesh height synchro

rodj42, 2022-02-11:
### Version
_2.3.3
MAC Monterey
Ender 3 V2 Marlin
 Why it's not possible (or rather very difficult  2.3.3) to get support/ model heights synchronized. I've used Cura/Ender3V2), but I would think that logic says if you have a hotend sweeping a surface it SHOULD have a level surface ,no matter how many extruders you have.

lukasmatena, 2022-02-11:
If you need support synchronized with object, Go to Print Settings ->Supports, set 'Top contact z distance' to 0, check 'Synchronize with object layers'.

I don't understand your reasoning about the level surface. The advantage of not synchronizing is that support layers can be thicker and faster to print.

If this does not solve your problem, post a 3MF and describe what you want to achieve.

rtyr, 2022-12-06:
Answered. Also a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7904. Closing.

closed 2022-12-06 by rtyr