prusa3d/PrusaSlicer #7905: Segfault when changing printer after loading 3mf with a different printer set

Thynix, 2022-02-11:
### Version

I encountered this in **2.4.1-beta** Linux64 GTK3 AppImage, and can reproduce it in **2.4.0** Linux64 GTK3 AppImage too.

### Operating system type + version

Ubuntu 21.10 x86_64

[system-info.txt](https://github.com/prusa3d/PrusaSlicer/files/8048655/system-info.txt)

### 3D printer brand / version + firmware version (if known)

I'm not interacting with the printer to trigger this bug, but it's a Creality Ender 3 Pro with Marlin 2.0.9.3.

### Behavior

1. _File > Open_ the attached _firstlayermatrix.3mf_ project.
2. From the _Printers_ menu, select **Creality Ender-3 BLTouch**.
3. Prusa Slicer immediately crashes.

Examples from the 2.4.1-beta1 Linux64 AppImage:

```
slic3r_main[123904]: segfault at ffffffffffffffe7 ip 0000000000d975d2 sp 00007ffefd4cbb70 error 5 in prusa-slicer[451000+1930000]
Code: e9 bb fe ff ff 48 89 c5 e9 4e 16 74 ff 90 66 2e 0f 1f 84 00 00 00 00 00 41 54 49 89 fc 55 48 83 ec 18 48 8b 36 48 8d 6c 24 08 <48> 8b 46 e8 48 89 ef 48 8d 14 86 e8 6e ee ff ff 48 8b 44 24 08 48
slic3r_main[124273]: segfault at ffffffffffffffe7 ip 0000000000d975d2 sp 00007ffdffdb1550 error 5 in prusa-slicer[451000+1930000]
Code: e9 bb fe ff ff 48 89 c5 e9 4e 16 74 ff 90 66 2e 0f 1f 84 00 00 00 00 00 41 54 49 89 fc 55 48 83 ec 18 48 8b 36 48 8d 6c 24 08 <48> 8b 46 e8 48 89 ef 48 8d 14 86 e8 6e ee ff ff 48 8b 44 24 08 48
slic3r_main[124761]: segfault at ffffffffffffffe7 ip 0000000000d975d2 sp 00007ffd6b55c6e0 error 5 in prusa-slicer[451000+1930000]
Code: e9 bb fe ff ff 48 89 c5 e9 4e 16 74 ff 90 66 2e 0f 1f 84 00 00 00 00 00 41 54 49 89 fc 55 48 83 ec 18 48 8b 36 48 8d 6c 24 08 <48> 8b 46 e8 48 89 ef 48 8d 14 86 e8 6e ee ff ff 48 8b 44 24 08 48
```

Examples from 2.4.0:

```
slic3r_main[127146]: segfault at ffffffffffffffe7 ip 0000000000d975d2 sp 00007fff3c643290 error 5 in prusa-slicer[451000+1930000]
Code: e9 bb fe ff ff 48 89 c5 e9 4e 16 74 ff 90 66 2e 0f 1f 84 00 00 00 00 00 41 54 49 89 fc 55 48 83 ec 18 48 8b 36 48 8d 6c 24 08 <48> 8b 46 e8 48 89 ef 48 8d 14 86 e8 6e ee ff ff 48 8b 44 24 08 48
slic3r_main[128140]: segfault at ffffffffffffffe7 ip 0000000000d975d2 sp 00007fffe32b9d10 error 5 in prusa-slicer[451000+1930000]
Code: e9 bb fe ff ff 48 89 c5 e9 4e 16 74 ff 90 66 2e 0f 1f 84 00 00 00 00 00 41 54 49 89 fc 55 48 83 ec 18 48 8b 36 48 8d 6c 24 08 <48> 8b 46 e8 48 89 ef 48 8d 14 86 e8 6e ee ff ff 48 8b 44 24 08 48
slic3r_main[128594]: segfault at ffffffffffffffe7 ip 0000000000d975d2 sp 00007ffdef40b770 error 5 in prusa-slicer[451000+1930000]
Code: e9 bb fe ff ff 48 89 c5 e9 4e 16 74 ff 90 66 2e 0f 1f 84 00 00 00 00 00 41 54 49 89 fc 55 48 83 ec 18 48 8b 36 48 8d 6c 24 08 <48> 8b 46 e8 48 89 ef 48 8d 14 86 e8 6e ee ff ff 48 8b 44 24 08 48
```

I got this from 2.4.0 a few times. It didn't happen every time, but when it did, the crash was not a segfault:

[image]

#### Project File (.3MF) where problem occurs

[firstlayermatrix.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/8048758/firstlayermatrix.3mf.gz)

bubnikv, 2022-02-11:
Maybe you are really low on RAM?

Thynix, 2022-02-11:
I don't think so. `free -m` reports over 1400 MiB available, and it doesn't happen if I import the project instead of opening it.

lukasmatena, 2022-02-15:
@Thynix Can you please post a complete config bundle (File->Export->Export Config Bundle)? I'm struggling to reproduce it, but it is possible that this is caused by something in your presets. Please, tell us exactly which printer is selected when PS starts and all steps leading to the crash. Then we may be able to reproduce it (although I doubt we will, to be honest).

If you sometimes get a bad_alloc and sometimes a segfault, I'd guess that it is not caused by memory exhaustion, but an actual heap corruption.

Thynix, 2022-02-15:
@lukasmatena Sure - here's [PrusaSlicer_config_bundle.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/8067580/PrusaSlicer_config_bundle.ini.zip).

When PS starts, _Original Prusa Mini & Mini+_ is selected. I don't know what more steps to add to what I already listed, so I took a screen capture in the hopes I'm missing something about what I'm doing:  [simplescreenrecorder-2022-02-15_04.23.46.zip](https://github.com/prusa3d/PrusaSlicer/files/8067667/simplescreenrecorder-2022-02-15_04.23.46.zip)

From that, steps I took:

1. Open PS
2. _File_ > _Recent Projects_ > _firstlayermatrix.m3f_
3. From the Printers menu, select **Creality Ender-3 BLTouch**.
4. PS crashes.

When I first went to reproduce it again with 2.4.0, for the first time it didn't crash, but then when I tried again with 2.4.1 it was back to

```
segfault at ffffffffffffffe7 ip 0000000000da7d92 sp 00007ffe9fe2ec20 error 5 in prusa-slicer[451000+1940000]
```

 When I was writing up this bug report I never had it not crash, so there's definitely something strange here.

lukasmatena, 2022-04-05:
Reproduced. You have to be in simple mode to get the crash. We will fix it.

lukasmatena, 2022-04-13:
Should be fixed in [PrusaSlicer 2.4.2-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2-rc2). Let us know if it isn't. Closing the issue.