prusa3d/PrusaSlicer #7906: PrusaSlicer 2.4.1 Beta 1 does not export in stl the dimensional changes made to the shapes.

liftbag, 2022-02-11:
### Version
_PrusaSlicer version 2.4.1-beta1+x64_

### Operating system type + version
_macOS Monterey Version 12.1_

### 3D printer brand / version + firmware version (if known)
_Not relevant_

### Behavior
* _Any dimensional changes made using the 'Scale' tool are ignored when exporting to stl._
* __Steps needed to reproduce the problem__
  * _Open PrusaSlicer 2.4.1-beta1_
  * _Add any shape, for example a box_
  * _Make a change, even asymmetrical, with the scale tool_
  * _Esport the modified shape as .stl_
  * _Clean the bed_
  * _Open the esported shape_
* __Expected Results__
  * _the imported shape should mirror the modified one_
* __Actual Results__
  * _The exported shape mirrors the original shape before being edited with the 'Scale' tool_

I have verified that previous versions of PrusaSlicer keep the changes made to the shapes when exporting to .stl

Jebtrix, 2022-02-12:
Yes it broke in 2.4.1-beta1 but its working again in latest git master.

lukasmatena, 2022-02-14:
Reproduced in 2.4.1-beta1, we will fix it. Thanks.

lukasmatena, 2022-02-17:
It should be fixed in 2.4.1-beta2 (14b7d36), let us know if problems persist. Closing.

closed 2022-02-17 by lukasmatena