prusa3d/PrusaSlicer #7908: Right layer slider display error

Sargastic, 2022-02-11:
### Version 2.4.0

### Operating system
Linux Gentoo

### Behavior
When slicing any model, the right-hand layer slider display is wrong (off by one layer).
Here is a (useless) model, 5 layers high, and playing with color changes to make the problem as plain as possible.

[image]

The first layer is in yellow, then two red layers, one yellow, and one green (with 2 walls, so it looks like there are 2 green layers).
The numbers on the right-hand slider are wrong :

- there are only 4 layers shown on the slider, the last one (green) does not appear
- the first (yellow) layer color is shown to go from 0.2 to 0.4 mm height
- all other layers are shifted up the same way

Looks somehow like issue https://github.com/prusa3d/PrusaSlicer/issues/4866

lukasmatena, 2022-02-14:
The conversation in issue #4866 that you mentioned ended after the user failed to provide data. Would you mind providing a 3MF here? I was unable to reproduce it in 2.4.0.

Sargastic, 2022-02-15:
> The conversation in issue #4866 that you mentioned ended after the user failed to provide data. Would you mind providing a 3MF here? I was unable to reproduce it in 2.4.0.

Of course, here it comes. I just want to emphasize the fact the problem is in the layers slider display, not in slicing or anything.

Sargastic, 2022-02-15:
Sorry about the closing, problem between chair and keyboard.
I have a 3MF ready, but how do I upload it ? I cannot attach it to the comment (filetype not recognized).

Sargastic, 2022-02-15:
[Test.zip](https://github.com/prusa3d/PrusaSlicer/files/8074207/Test.zip)

Okay, maybe not the best way but... Here is a 3MF file, renamed to ZIP extension for the upload to be allowed. It's a (useless) 5 layer slab, just to show the problem. In the « sliced » view, I have already added colors (fourth layer is blue instead of yellow as in the first message).

closed 2026-09-29 by prusaslicer-bot[bot]