prusa3d/PrusaSlicer #7910: hey so when i put in my stl file or whatever my screen goes black like the little view screen goes black and then when i put my cusor over it it flashes, i can still slice and evrything but i dont know whats up with the slicer, Im running it off chromebook btw

ryantheman104, 2022-02-12:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Jebtrix, 2022-02-12:
What method did you install Prusaslicer with? Flatpak, appimage, or some debian package? What make/model is the chromebook? I've seen @bubnikv stating recent efforts to get this working more smoothly: https://forum.prusaprinters.org/forum/prusaslicer/how-to-install-and-run-prusaslicer-on-chromebook-with-linux-support/

tliguori, 2022-02-13:
I am having the same issues running on windows 11 on a laptop. Blinking black screen and freezing only after slicing. My laptop seems to be fighting with the graphics

ryantheman104, 2022-02-13:
> What method did you install Prusaslicer with? Flatpak, appimage, or some debian package? What make/model is the chromebook? I've seen @bubnikv stating recent efforts to get this working more [...]

i have i used the appimage

lukasmatena, 2022-02-15:
@ryantheman104 I have edited your post and moved the text from the title to the body and removed the issue template that you left blank. We will try to reproduce your issue. Does it happen with any model?

bubnikv, 2022-02-15:
What model of Chromebook? What CPU / graphics card? We only have access to three chromebooks, one intel and two ARM. The more powerful ARM Chromebook has issues when switching Linux applications, that may sometimes kill PrusaSlicer. I suppose it is a specific issue to that particular Chromebook Linux OpenGL implementation.

Ardustorm, 2022-05-14:
I helped my dad with a similar issue. We were able to resolve it by disabling what I believe is the gpu acceleration support (the `crostini-gpu-support` flag).  That makes me think it probably isn't a PrusaSlicer specific issue.

If anyone runs into the same issue, here is the thread I referred to know which flag to disable:

https://support.google.com/chromebook/thread/41107283/many-linux-apps-can-only-render-images-as-black-rectangles?hl=en

bubnikv, 2022-05-16:
Thanks. It is best to report your particular chromebook model and chromeos
version directly to Google.

lukasmatena, 2023-03-27:
Is it still happening in 2.6.0-alpha5? Thanks.

closed 2026-09-29 by prusaslicer-bot[bot]