prusa3d/PrusaSlicer #7912: weird behavior after model cutting

coufy, 2022-02-12:
### Version
2.4.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa Mini

### Behavior
When I import the model (.stl) and slice it everything is OK. When I cut the model in 4mm height and slice it, PS "inverts" the model and fills empty volumes and does not fill the real volumes of the model.  STL and screens included.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]
[image]
[image]
[image]
[Eight v2 - 3DP (2).zip](https://github.com/prusa3d/PrusaSlicer/files/8054333/Eight.v2.-.3DP.2.zip)

Jebtrix, 2022-02-13:
Looks to be fixed and working fine in [version_2.4.1-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.1-beta1)

lukasmatena, 2022-02-15:
I don't think this is fixed. The cut function is not robust enough and may produce broken models in some special cases. Cutting at 4.01 mm is already fine based on my experiments. We should try to improve it one day.

closed 2026-09-29 by prusaslicer-bot[bot]