prusa3d/PrusaSlicer #7915: object moves off build plate when slicing

flyingnunley, 2022-02-13:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3
[Body44.zip](https://github.com/prusa3d/PrusaSlicer/files/8056152/Body44.zip)
MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

I have been trying to use 3mf files but lately every 3mf file I have tried, the object moves off the build plate when I slice it.  If I use an stl of the same model, it works fine.

neophyl, 2022-02-13:
Working fine for me.  PS2.4.1 beta on Windows 10.  Opened up your project, the object is in the middle of the build plate and sliced, object is still there and location matches.

You haven't mentioned what version you are running or what OS etc.

The coordinates of your object are very offset from 0,0 (x1642, z -3056) and iirc there were changes to cope with 3mf placement when imported as the global positioning from your cad software was respected when imported previously.  I think those changes were in the normal 2.4 release.

flyingnunley, 2022-02-13:
> Working fine for me. PS2.4.1 beta on Windows 10. Opened up your project, the object is in the middle of the build plate and sliced, object is still there and location matches.
>
> You haven't mentioned [...]

Thanks for the response.  Your mention of the offset got me thinking.  I don't know much about the 3mf format but I assume it contains some data about the coordinates of the object in space.  I was saving the object from a small body within a very large model in fusion 360.  Apparently, fusion 360 saves the body's coordinates in the 3mf file.  As an experiment, I moved the body in fusion 360 to the origin, then resaved the body as a 3mf and sure enough, it landed right on the build plate in Prusaslicer.  I understand now why it does this but its kind of annoying.  I don't see a way in either program to reset the origin of the 3mf file.

manunle, 2022-02-14:
> 

lukasmatena
Yes I am using 2.3.1.  Its interesting because when I first load the file, it ends up way off the build plate.  I then move it to the build plate, slice it and bam, its way off again and it displays an error.  When I get a chance, I'll install 2.4.1-beta1 and see if it works.

lukasmatena, 2023-02-06:
Should be fixed in 2.4.1 and later. Closing.

closed 2023-02-06 by lukasmatena