prusa3d/PrusaSlicer #7918: Changing setting with Edit model/object

Nj-BigMike, 2022-02-14:
### Version
Version 2.4.0+Win64
Windows 10 Pro
TronXY X3A Marlin 2.0.9

While slicing I use the edit feature on a model to adjust settings. When I select Skirt and Brim (skirt is already on by default) I see no option to remove skirt. It would be nice to be able to do this when dealing with a print that after the brim goes right to the edge.

lukasmatena, 2022-02-14:
The skirt is a property of the whole print, not of an object. That's why it cannot be overridden in object settings (unlike brim, which can be).

> when dealing with a print that after the brim goes right to the edge.

I don't understand the use case.

foreachthing, 2022-02-14:
> I don't understand the use case.

Big prints, close to the edge *and* skirt can lead to a skirt outside the print region. Happens to me very often. 

@Nj-BigMike just reduce space between skirt and model or simply disable it. Prime the nozzle in start gcode.

Nj-BigMike, 2022-02-14:
> @foreachthing <https://github.com/foreachthing> Thanks, I get it now.
>
> We do not plan to change this behavior in near future, having multiple
> skirts would probably create more problems that it would fix. Either turn
> the skirt off or print it set as 'draft shield'.

I wasn't after multiple skirts -- just the ability to turn it off without
changing the preset when it becomes visible that the skirt won't fit.

thanks for the response.
Mike

Nj-BigMike, 2022-02-14:
That would be ok, I was just looking to be able to do it without going back
and changing the preset thats all.

foreachthing, 2022-02-14:
> Then perhaps the menu item should just say Brim rather than Skirt and Brim that implies an ability to change the skirt option. 

You can change Skirt settings. Maybe you need to enable "advanced user" first!?

Nj-BigMike, 2022-02-14:
Other posts say I can't change Skirt option since it is a print setting not
an object setting.
I'm in expert mode. I am referring the the edit menu when you click on an
object to adjust layers,infill etc.

foreachthing, 2022-02-14:
Now **I** understand. A screenshot would've helped me ;-)
You're "here", right?
[image]

You'd have to disable Skirt in print settings => set loops to zero. Or decrease distance.

neophyl, 2022-02-14:
Technically as it doesn’t contain skirt then yes Brim would be a better title, however I thought it just took those headings from the section the setting resides in. As those settings are in a section called skirt and brim the names all match up and you know where to go looking for equivalent setting.

lukasmatena, 2022-02-15:
> it just took those headings from the section the setting resides in

Exactly. The caption is automatically taken from the section name. And although it is possible to make this a special case, I'd rather leave it as it is for consistency reasons. It means "show me the options from 'skirt and brim' category that I can override per object".

Nj-BigMike, 2022-02-15:
Fair enough.
I appreciate the response.

closed 2024-02-19 by lukasmatena