prusa3d/PrusaSlicer #7919: This is more of a feature request than a bug

Nj-BigMike, 2022-02-14:
I'm using ver 2.4.0+win64 on windows 10 Pro

Since as we all know when we change print speeds from say 20mm/s to 100mm/s we sometimes need to  increase or decrease the heat setting for the filament.
[image]
[image]

It would be great if filament settings could be such that
N to 40 is temp1
41- 80 is temp 2
based on max print speed of course.
Either programaticaly or by allowing an edit to filament temperature when editing an object.

My apologies if this is the wrong location for this request.

lukasmatena, 2022-02-15:
> based on max print speed of course.

max_print_speed has little to do with the actual speeds that are used. And these typically change quite often, the extruder temperature would hardly be able to follow the changes.

Do you have a real world use case where you have actually post-processed some G-Code and got better results? What was it? Or do you just assume that this will improve things?

Some magic of this sort should be doable e.g. in layer change G-Codes. Also, it kind of looks like a duplicate of #6228, if you agree, I'd close this issue and discussed in the other one.

Nj-BigMike, 2022-02-15:
Sorry I wasn't clear enough apparently.

Looking for a ONE TIME adjustment to the temperature based on my max speed.
If I print at 20mm/s I run PLA at say 205C when I print at 80mm/s I need to
bump that temp to 215C or my extruder can't keep up[.
I'm not asking for a dynamic temperature change that would be quite
impractical.
Just a one time adjustment at the start of the print to the temperature
either manually set with an edit feature like the object edit but for the
whole print or a setting to reference in filament that is read along with
max speed when slicing and decides SLow 205, medium 210, fast 215

Is this clearer for what I meant.

foreachthing, 2022-02-16:
Maybe you find something here: https://help.prusa3d.com/en/article/macros_1775/.

closed 2026-09-29 by prusaslicer-bot[bot]