prusa3d/PrusaSlicer #7920: [Feature request] Feature to specify ironing angle

TKosir, 2022-02-14:
### Version
2.4

### Operating system type + version
Windows

### 3D printer brand / version + firmware version (if known)
MK3s 
E3D toolchanger (RRF 2.x)

### Behavior
It would be very useful if one could specify the angle of ironing regardless of the infill angle beneath.
Additionaly it would be very useful to make ironing a separate function, that would add ironing as a modifier. That way only specific parts of the model could be ironed with ease +  multiple parts of different materials (multimaterial printing) could be ironed at once. 

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
None

rtyr, 2022-02-15:
Seems like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5411.

TKosir, 2022-02-16:
It is not he same as #5411, since that issue can be solved to change the layer infill angle under the ironing. It would be really beneficial in my situation of I could perform ironing at angles 0 and 90 degrees relative to the infill angle.

koisdigala, 2022-10-29:
Additionally, I'm not sure a fixed angle is best - some surface shapes don't care. But as an example, a long thin area to iron looks terrible if you iron it in any direction except parallel to the long axis. I have items which get iron perpendicular to the long axis and it looks worse than the original top surface. So some intelligence requested!

So consider a box where the tops of each wall are wide enough to iron (say 4mm). Different tops need to be ironed at different angles.

lukasradek, 2023-01-22:
I would also add that it would be useful to always / on demand make the ironing passes along the long axis. Eg. if you have rectangular area 2x100mm and "stars" align incorrectly with the current algorithm you can end up making very many 2mm passes along the short dimension instead of a few long 100mm ones along the long dimension.

annevanleyden, 2023-06-09:
+1 on the ironing in longitudinal direction.

pavolNovak, 2023-10-26:
+1 on specifiing angle direction on ironing.

closed 2026-09-29 by prusaslicer-bot[bot]