prusa3d/PrusaSlicer #7922: Prusa Slicer

NICENYANCAT, 2022-02-14:
Prusa Slicer won't open despite any version I try to use.
[image]

foreachthing, 2022-02-14:
Hit [CTRL]+[C] while this window is open (paste to notepad) and you might get the full content of it and can remove/edit the faulty line in your PrusaSlicer.ini.

bubnikv, 2022-02-15:
@[NICENYANCAT](https://github.com/NICENYANCAT)

Your PrusaSlicer.ini is corrupted. It happens on Windows rarely and since PrusaSlicer 2.3.3 we are creating back-up of PrusaSlicer.ini, so you should no more get such error message.

Which version of PrusaSlicer are you running? How did you get into such situation?

NICENYANCAT, 2022-02-19:
> 

I have tried 2.4.0 and 2.3.3 and both of them showed an error message. I do not know how this happened.

foreachthing, 2022-02-19:
Try to fix your ini. 
If you can't, zip it an post it here, maybe I can help (or someone else)...

NICENYANCAT, 2022-02-19:
> Try to fix your ini. If you can't, zip it an post it here, maybe I can help (or someone else)...

I'm not sure how to do that.

foreachthing, 2022-02-20:
Which part? 
You can find your prusaslucer.ini at the location from the error message. Navigate there. Rightclick that file and zip it. Then attach that zip file here (drag and drop).

closed 2026-09-29 by prusaslicer-bot[bot]