prusa3d/PrusaSlicer #7925: What triggers the MMU parameters to take effect in slicing?

LightningShark25, 2022-02-16:
### Version
2.4.0

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 with custom 2 into 1 Y extruder

### Behavior
Took my single extruder profile and copied it.
Added a second extruder
Ticked the box 'Single Extruder Multi Material'

Enabled Wipe Tower
Tuned retraction distances through MMU settings etc

I have not added any custom gcode for this mod.

I was expecting that the settings would handle the retraction of filament for changeover etc
Also not seeing any evidence of ramming profiles being used
Or purge volumes.. 
[image]

What am I doing wrong please?  Or is there a bug somewhere?

_Is this a new feature request?_
Not at this stage
[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/8076624/Shape-Box.zip)

lukasmatena, 2022-02-16:
I don't see any issue. The wipe tower is generated, first tool is rammed on the first layer, tool is changed, the other tool is wiped according to purge volume settings, then the other half of the model is printed. Transition to the second layer, ramming of the loaded filament, unload, end. What is the problem?

LightningShark25, 2022-02-16:
This might be me not understanding what prusaslicer is trying to generate in GCODE.

From what I've seen the printer do...:
There is no retraction of filament before tool change, or any priming of filament on the new tool - hence I'm not sure what the following settings are defined for:
[image]

I don't believe I observer ramming of filament - unless this is embedded in the wipe tower deposition? 

Does the purge volumes in the dialog dictate how much filament is deposited in the wipe tower.  My assumption that the wipe tower volume of filament would be determined by its size  - specified in the wipe tower settings.

I have manually worked around a lot of this with custom g-code - but I guess I was expecting from MMU settings - that these were taken care of by the slicer?

Feel free to correct me if I've not got this right!
Thanks

lukasmatena, 2022-02-16:
You have finished with the old filament, travel to wipe tower. Ramming (the thick lines of the old filament on the wipe tower). Retract to cooling tube, cool the filament in the cooling tube by moving it back and forth a bit. Then retract to parking position. Change tool (T1). It is now in parking position, so unretract it to where the nozzle should be. Extra unretract according to "extra loading distance". Start wiping the new filament on the wipe tower until you extrude the value from purge volumes. You can now print with the new filament.

> I don't believe I observer ramming of filament - unless this is embedded in the wipe tower deposition?

It is on the wipe tower.

> Does the purge volumes in the dialog dictate how much filament is deposited in the wipe tower. My assumption that the wipe tower volume of filament would be determined by its size - specified in the wipe tower settings.

Higher purge volumes mean larger wipe tower, yes.

LightningShark25, 2022-02-16:
Thanks for your responses.

I just found an issue in my octoprint config that was resetting my extruder back to absolute position...  This will have definately upset my extrusions - and probably explains why my extruder didn't do what I was expecting.  Needed an extra M83 to sort that :(

So.... will report back when I retest - printer in bits right now.

lukasmatena, 2022-02-16:
Ok. I'm closing the issue, I don't think this is a problem in slicer. It is still possible to comment on it (and it can be reopened in case it turns out to be a bug in the end).

closed 2022-02-16 by lukasmatena