prusa3d/PrusaSlicer #7926: Bridge lines rendered incorrectly in layer view

jmding8, 2022-02-16:
2.4.0, test object sliced with `0.20mm DETAIL @0.6 nozzle MK3` profile (which has thick bridges disabled)

For reference, View = Feature Type. You can see the bridge infill clearly.

[image]

Switching to View = Height, we can see that the bridge infill is rendered with 0.20mm height

[image]

Switching to View = Width, we can see that the bridge infill is rendered with 0.66mm width

[image]

However, we know that unsupported extrusions in free air have circular cross section. This means that the bridge infill should be rendered with height = width = 0.40mm (roughly).

Assuming we agree on the above, it implies a deeper underlying issue: the way we currently configure bridging settings produces bridge layers that are thicker than modeled. However for now, let's just focus on the fact that the gcode visualizer is not handling bridges correctly.

rtyr, 2022-02-16:
Intended behavior with "thick bridges" disabled. See https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1.

> Starting with this release, PrusaSlicer newly offers printing bridges and 1st object layers over supports with the default layer thickness. The bridging behavior is configurable with a new "thick_bridges" boolean option. If enabled, the old Slic3r bridging strategy is applied, while if disabled, bridging is performed in a similar fashion to the other FDM slicers, leading to better bottom quality over supports. With "thick_bridges" disabled, the behavior of "bridge_flow_ratio" changes: Instead of modifying cross section of a rounded extrusion while keeping the gap between the extrusion threads constant, extrusion spacing is maintained while modifying the extrusion width and / or height to reach the prescribed extrusion ratio.

jmding8, 2022-02-16:
I understand that this is intended behavior for gcode generation. However it is clear to me that this should not be intended behavior for the gcode previewer. Unsupported extrusions in free air (i.e. bridge lines) have circular cross section. And thus they should be previewed with circular lines, rather than flat-oval lines.

rtyr, 2022-02-16:
The preview is created from the g-code.

Intended behavior, closing.

closed 2022-02-16 by rtyr

jmding8, 2022-02-16:
I give up. Y'all are clearly not willing to even consider feedback given with good intentions.

jmding8, 2022-02-16:
The gcode sets the width, which the previewer consumes

[image]

The gcode sets the wrong width. For bridge lines, the width MUST be equal to the height because they are extruded in free air