prusa3d/PrusaSlicer #7929: Add „depth“ option for multimaterial painting

michael-kerbel, 2022-02-16:
When painting in multimaterial, it is pretty important for the result, how deep the painted color goes. Dependent on the color combination, the color below will bleed through. That can actually be used to get way more colors than the MMU allows, as shown by a Reddit user:

https://reddit.com/r/3Dprinting/comments/s6f3wj/pro_tip_you_can_add_shading_to_your_multimaterial/

[image]

That‘s why I would love to have a „depth“ option when painting multi material. That way it would be possible to finetune the resulting colors.

wsmagee, 2022-09-10:
This could also allow for some fine tuning to minimize color changes on the top layers of certain objects.

DragonFinlay, 2022-09-20:
Thanks for sharing, I really love this idea, it kinda blows my mind. I assume it would be pretty time-consuming to implement, but I'll make sure the team sees this!

ssjbardock, 2022-09-25:
Plus keep in mind a lower layer height could provide a finer gradient of shading, while adding more undertone layers could widen the range. 
Could produce nearly and shade between a set of colors!
Even with the base '5 shades per color base', that comes to 20 colors with a MMU2S! (base colors full swatch, Black/White share a gradient)

ftheriachab, 2022-10-01:
> 

Hi, original poster of the linked reddit post.  You would think you get get a finer gradient with a smaller layer height, but in practice flow control really becomes an issue.  You end up seeing striations across the print rather than an even coat.

DragonFinlay, 2022-10-04:
Yep, can confirm. I experimented with 0.1-0.3mm layers and there's not much difference. From what I saw, 0.2mm layers come out best.

toddanglin, 2022-11-16:
This would be a really helpful feature for fine tuning multi-material prints even just to get cleaner results.

I often have the desire to begin a multi-material transition several layers lower than default so there is a cleaner transition at material boundaries. This is especially true on prints with flat surfaces at different heights.

So big +1 to adding some configurability to layer control for painted multi-material prints.

mepoland, 2023-01-09:
I'd like this feature in order to write 1-layer thick text in a different color on top of a part, like this:
[image]
Currently I model the text as 1 layer thick, but when I paint the top surface of text black, I end up with several unnecessary black layers going down into the part. Number of layers = number of "top" layers in print settings for the part. This messes with the part structure and causing extra material changes:
[image]
My workaround is to split the embossed text into a different part with a different color.

michael-kerbel, 2023-01-28:
Here is another example of a user playing around with this idea:
https://www.printables.com/de/model/375015-3-color-blend-swatch

jpf11, 2023-11-11:
Would really like this feature, because having colors bleed through the single layer of another color is really annoying.

phat3r, 2023-11-30:
I also support this request.  

When making signs I need to paint the faces AND ALL THE EDGES of the raised text ... if it had a depth option then just the face would suffice.  Currently painting all the raised faces AND their edges  (I set the smart fill angle to 89 degrees) takes 3 times longer than just the faces alone and it's easy to miss one small part of the text when you have 40-50 characters you need to paint all around.

The current bleed down in layers also ruins shallow prints where you want and only need one layer of colour for the text.

I don't believe it's possible to even mix the layer based colours with the painting tool.

There is definitely room for massive improvement in Prusa Slicer around colour.

digi2life, 2023-12-09:
@phat3r I just ran into the same issue with the painting overriding the layer based filament/extruder changes. I'm using the XL.
We definitely need more control of the painting transition into adjacent areas.

jurassic73, 2024-04-24:
Any update or progress on this? I want to print a sign with transparency and this bleed through to lower layers is messing up the colors.

dgoldenholz, 2024-07-03:
Agree that control over depth would be helpful.

Flyingvette, 2024-09-07:
Please devs - this is a major issue that has affected hundreds of my prints. I have multiples right now that won't print correctly at all due to being unable to control the top x layers of color / multimaterial deposition.

Electrode927, 2024-09-29:
This capability would be valuable in a number of circumstances. I just ran into a situation where negative extrusion from a face painted differently from the default color of the object presented a nice aesthetic, but I had no choice to control the depth of that paint. 

Seems like a good feature.

Badfurball, 2025-03-19:
It really frustrates me that I can't paint an area of the print and have that color go all the way through it. Like glowing eyes on a print that is illuminated from the inside with a translucent filament???

TheIvoryDisaster, 2025-06-20:
Please work on this. It's essential for multi color printing. So many filaments that have colorful additives (which are becoming more and more popular) do not look good at all with the current multimaterial settings available.

pmperry, 2025-06-25:
If you use smart fill you can choose the depth with that.

michael-kerbel, 2025-06-26:
@pmperry Really? Where? Please show us! I could not find anything.

pmperry, 2025-06-27:
@michael-kerbel reduce the degrees and it reduces the paint depth as well.  It’s not perfect, but it can be done.

TheEvilRocky, 2025-07-24:
Agree, this needs to be implemented, makes far more easier to paint and get the desired result!

djmeph, 2025-08-24:
I would like to see this feature because I would like to be able to Smart Fill between edges and set the depth to go all the way through to the other edge, so I can create a window with clear filament.

Remklep, 2025-12-27:
I would like this feature very much as well!

michael-kerbel, 2026-09-04:
Since it is not implemented in 3.0 and I still believe, it would be a very useful feature, I would like you to keep this feature request open.

jamesra, 2026-09-04:
I concur with the request to keep this feature open.

Jan-Soustruznik, 2026-10-19:
Hello, thank you for the update. Following our new [policy,](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I am closing this feature request and converting it into a Discussion thread. This will allow us to hear what the community thinks about this feature and provide a place for others to upvote it. Have a great day.