prusa3d/PrusaSlicer #7932: Graceful end of vase mode print

firepebble, 2022-02-17:
### Version
_2.4.0_

### Operating system type + version
All

### 3D printer brand / version + firmware version (if known)
All

### Behavior
* When printing in vase mode the model will finish suddenly when the height is reached. This causes a jagged top at the end point and overall top is not flat.
* It would be better to allow variable height on the top of a vase mode print so that the end point is less jagged and the top is more flat. This would result in a much nice finish.
* A flatter top would also allow ironing to be turned on for "Topmost Surface" in vase mode.
* 
_Is this a new feature request?_

Yes.

gaduffl, 2022-02-27:
100% agree, makes vase mode basically unusable when working with large layer height.

Zaaijer, 2022-03-05:
I've was wondering the same thing without finding a solution of my own. I regularly print vases with large layer heights, and the lack of this feature makes me prefer Cura over PrusaSlicer. Would love to see this feature be added in the future!

firepebble, 2022-03-11:
If anyone is doing this manually here is what I found works on bottle tops that have a fairly small circumference and 64 'sides'. Find the last layer in the g-code, which might look like this:

G1 Z72.301 X134.438 Y82.347 E.03928
G1 Z72.308 X134.722 Y83.314 E.16997
...
G1 Z72.683 X133.043 Y79.675 E.16983
G1 Z72.689 X133.595 Y80.519 E.17004
G1 Z72.695 X134.062 Y81.412 E.16996
G1 Z72.700 X134.351 Y82.13 E.13065

And use a spreadsheet (see attached [Vase top gcode modification.ods](https://github.com/prusa3d/PrusaSlicer/files/8236372/Vase.top.gcode.modification.ods)) to make new values to add, which go around one extra turn using the extraction rate from the 62nd 'side' and phase down the extraction linearly down to a negative value right at the end:

G1 Z72.700 X134.438 Y82.347 E0.1672
G1 Z72.700 X134.722 Y83.314 E0.1645
...
G1 Z72.700 X133.043 Y79.675 E0
G1 Z72.700 X133.595 Y80.519 E-0.0027
G1 Z72.700 X134.062 Y81.412 E-0.0055
G1 Z72.700 X134.351 Y82.13 E-0.0082

Clear as mud right? I'm sure others can do a much better job ;-)

Zaaijer, 2022-03-13:
Thanks, @firepebble!

I had a look at the ending of the vase mode when slicing with Cura and it's doing the exact same as you describe, but automatically. Just started a test print to see if I can implement it manually. I regularly print using vase mode, so integrating it in the slicer itself would still save me some time though :D

clivius, 2022-06-24:
It would also be helpful if the print ended with a wipe in the opposite direction of travel so as to not leave a string of material from the sudden end and lift to the park position.

firepebble, 2022-07-20:
@clivius I find it works better to not have a wipe move but to start retracting as you get near the end of the extra turn, then do a sudden larger retraction with an inwards move to 'snap' the filament at the end.

Misfi, 2022-09-17:
👍 
Would love to see it. Using a 1.8 mm CHT nozzle and 1.5 layer height, it just... doesn't look good :-)

jacoblukewood, 2022-11-21:
If this were to be implemented, should there be a toggle, or should it just be the default action?

It almost seems like this should be a bug or oversight rather than a feature request; especially seeing as Cura leaves a clean end in vase mode.

firepebble, 2022-11-21:
Should just be the default to finish smoothly I would think. If it was a
toggle, maybe we would have to turn off smooth finish, but I honestly don't
know why a jagged edge would be useful.

b3n3d1k7, 2023-09-10:
I created a pull request to fix this issue: https://github.com/prusa3d/PrusaSlicer/pull/11282
It includes a new option to disable the fix and revert to the old logic:
[image]

There is still an issue with the preview of the last layer and I haven't done any test-prints yet.
Please let me know if you see any other problems.

Flush finish disabled:
[image]
enabled:
[image]

b3n3d1k7, 2023-09-17:
Preview now also works.
[image]
Just need to do some test prints.

firepebble, 2023-09-18:
Amazing work!

b3n3d1k7, 2023-09-20:
Had to fix a small bug I discovered during the first test-print, but now it works.
Here are two prints of the same cylinder with a 0.8mm Nozzle and 0.7mm layer-height.
The one on the left was printed with the fix on, the one on the right with the fix off - obviously :wink: 
[image]
The extrusion-ends right next to each other:
[image]
Here you can see the last layer with reducing extrusion to create a flush top layer:
[image]

b3n3d1k7, 2023-11-25:
Does anymore know what i got to do to get my PR reviewed?
It's been two months now and nothing happened - except my PR now having conflicts.
If this won't get reviewed anyways there is no point in fixing the conflicts ...

5jvm0u4, 2023-11-25:
No one really knows, I've been here for over a year, there's been nothing.

firepebble, 2023-11-26:
I wish this would get resolved - I'm still manually calculating the last
layer and fixing the gcode but how will we do this with the new binary
format being used?

MichaelJLew, 2023-11-26:
> Because of the way vase mode is implemented, it was quite simple to add this for vase mode. I assume it will be quite a bit more complicated to add this to non-vase mode slicing.

A bit more complicated, but really not much more. If you look at the example file that I put up in #11621 you will see that it has two perimeters. I described the moves needed too.

b3n3d1k7, 2023-11-27:
> A bit more complicated, but really not much more. If you look at the example file that I put up in #11621 you will see that it has two perimeters. I described the moves needed too.

I'm not talking about the gcode being more complicated, but how you'd have to change the c++ code generating the gcode to achieve the result 😉

MichaelJLew, 2023-11-27:
> how you'd have to change the c++ code generating the gcode to achieve the result

Yeah, I understand that. I can't do C++, but in my 'native' language, LiveCode, I wrote a short function for the tapered extrusions (20 lines of code) and another function that deals with the seam.

I think that the tapered extrusions function should be in the firmware rather than the slicer as it has utility for lots of non-planar tasks as well as the scarf seam and the flat top for vase-mode prints.

The slicer would need to keep track of the first lot of xy coordinates for the outer perimeter until the total movement distance reaches or exceeds a pre-set value and taper up over those points, and then append those same points to the end of the perimeter with the extrusion values tapering downwards to zero. Not hard at all in pseudo-code I think. Hard in C++? I don't know.

Tupson444, 2026-09-14:
still relevant

MichaelJLew, 2026-09-14:
Still relevant if it has not been fixed (I have not checked, but assume that @Tupson444 has done so). The code for the second part of the scarf seam will serve for this purpose so it should be pretty simple to implement...

Tupson444, 2026-09-15:
I didn't check this one (there are so many issues scheduled for closing unless someone replies), i just assumed that it wasn't implemented as I don't remember it being mentioned in release notes. But now I checked and it actually seems to be implemented.

firepebble, 2026-09-15:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

When I checked the implementation results in the new vasemode they were
substandard. They had adopted an algorithm from another codebase that had
variable layer height and the whole slicing of vase mode is terrible now
fpr fine detail and protrusions but it does a slightly better top layer. If
you slice in the old method it still has the rough top. But at this point
we would have to open a whole new issue under the new vasemode code.

Misfi, 2026-09-17:
FYI this issue will be closed since it's not a bug. You'll need to re-open this as Discussion.