prusa3d/PrusaSlicer #7933: upload gcode stack at 96% RepRap Duet + ArcWelder plugin

KRUP85, 2022-02-17:
### Version
`PrusaSlicer
Version:   2.4.1-beta1+win64
Build:     PrusaSlicer-2.4.1-beta1+win64-202202010918

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (sestavení 19044), 64bitová edice
Total RAM size [MB]: 17,042MB
OpenGL installation
GL version:   4.6.0 NVIDIA 457.49
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2060/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
[... 746 lines cut ...]
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

### 3D printer brand / version + firmware version (if known)
Duet3 6HC, RepRap 2.4 RC1 + SBC RPi 4

### Behavior
When sending gcode to the printer, the program always stops at 96%. However, Gcode is loaded in its entirety and fully functional. I use the ArcWelder plugin in post process scripts. When I don't use it, everything works without problems.

[image]

#### Project File (.3MF) where problem occurs

[fixa_cidlo_send.zip](https://github.com/prusa3d/PrusaSlicer/files/8089506/fixa_cidlo_send.zip)

StephenRC, 2022-02-17:
Duet 3 6HC+1LC+PI4, currently on 3.4.0.rc1, has been occurring on prior 
firmware versions.  Using arcwelder.  With the PI, upload never 
finishes, but in standalone the upload finishes.

-- 
This email has been checked for viruses by AVG.
https://www.avg.com

StephenRC, 2022-02-27:
In SBC mode with 2.4.1beta3, uploads never finish.

StephenRC, 2022-04-26:
Still never finishes uploading with rrf 3.4.0

StephenRC, 2022-06-18:
Still the same with 3.4.1 and PrusaSlicer 2.4.2

KRUP85, 2023-03-08:
Work without problem with 3.5.0 beta2 and PrusaSlicer 2.5.0.

KRUP85, 2023-12-06:
> Is this still an issue in 2.7.0?

No issue in 2.7.0. upload finish with no problem.

lukasmatena, 2023-12-06:
Thanks for the update. I guess we don't have to look for what change has fixed it. Closing.

closed 2023-12-06 by lukasmatena