prusa3d/PrusaSlicer #7934: Issue moving program from one screen to another with a different resolution

sworcester001, 2022-02-17:
### Version
2.3.0+wi64

### Operating system type + version
Win 10 Pro 21H2 build 19099.1526

### 3D printer brand / version + firmware version (if known)
MK3+S with Bondtech

### Behavior
I have one screen that is 4K and another next to it is 1080. If I open Slic3r on one screen and using the mouse to grab and pull/move it to the other screen, it freezes half way and then when/if it recovers (starts responding) I have to complete the pull. 
If the screens are the same resolution, the problem doesn't exist.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

foreachthing, 2022-02-17:
Do you have different DPI settings per screen?
PS 2.5.0a0 has no issues on my screens (1080 and 1200 height). Same DPI, though!

What happens if you use [WINDOWS]+[SHIFT] and [<-]/[->] (left or right cursor)?

sworcester001, 2022-02-17:
> Do you have different DPI settings per screen? PS 2.5.0a0 has no issues on my screens (1080 and 1200 height). Same DPI, though!

Different resolutions, yes. (I assume that's what you meant).  Monitor one is 3840x2160 and monitor 2 is 1920x1080.
If I change the resolution on monitor to 1920x1080, the problem is gone.

> What happens if you use [WINDOWS]+[SHIFT] and [<-]/[->] (left or right cursor)?

Windows <SHIFT> right cursor moves the whole screen but still puts it in to a non-responsive mode temporarily.

foreachthing, 2022-02-18:
With DPI I meant "Scale":
[image]

I could imagine, that the 4k has a 150% scale and the 1080 has 100%. Then, the application has to rescale while dragging and that, IMHO, could cause the delay. Two monitors, same scale (and resolution) the application does not require to rescale. This is my best guess ;-)

foreachthing, 2022-02-18:
Yeah, I just tried it at work (two 4k monitors) one set to 1080 and 100% scale, one at 4k and 150% -> GUI delay due to "redraw". I'd expect that and I think this is somewhat normal.... I don't see an issue here.

foreachthing, 2022-02-18:
I guess a windows application (wpf or uwp) performs much faster than wxwidgets... I am not a gui-person, so someone from PS might explain things?!

Samoth47, 2023-10-23:
Responding to this old issue, I have a similar setup (one 4K screen @ 1.5x scaling and two 1080p screens @ 1x scaling either side) and PS (2.6.1) keeps on crashing when a moderately complex model is loaded and the window is being moved from one screen to another. It also regularly crashes when the PC comes out from sleep, because each screen is waking up and Windows is constantly moving windows around different DPI screens (yes, it's a really frustrating experience).
It seems to crash whenever the app hangs rescaling for too long, like if a watchdog is timing out.

closed 2026-09-29 by prusaslicer-bot[bot]