prusa3d/PrusaSlicer #7935: PrusaSlicer is filling certain open areas of my model

KoenVanduffel, 2022-02-17:
### Version
2.4.0 but it does it in all versions I tested including 2.4.1 beta

### Operating system type + version
Win 10 21H1
PrusaSlicer
Version:   2.4.0+win64
Build:     PrusaSlicer-2.4.0+win64-202112211614
Full PrusaSlicer details below.

### 3D printer brand / version + firmware version (if known)
Creality CR6 SE running Klipper

### Behavior
* My model has a fan-shroud with stator blades. PrusaSlicer as well as SuperSlicer (partially the level differs per version the latest beta to fill them completely) fill the voids with solid infill.
Download the model from: https://www.prusaprinters.org/prints/126838-cr-6-direct-drive-dragon-hf-sf-and-orbiter-v15-mou/files and use the Dragon_&_Orbiter_mount.stl or 3mf files, the 3mf files gives even more of the artefact than the stl.

* any settings I tried all produce the exact same result. only verying the layer height can make the extent of the artefact different but I have not seen a single slice that went well.
model:
[image]
Slicer result:
[image]
Cura slicer result using very similar settings:
[image]

I tested the slicing of my model in PrusaSlicer after @Merafot commented on Discord that he had this problem: https://discord.com/channels/759603746270609428/760241486649491466/943866753056927804

_Is this a new feature request?_ NO

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 34,281MB
OpenGL installation
GL version:   4.6.0 NVIDIA 471.41
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GT 1030/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
[... 337 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

Jebtrix, 2022-02-18:
Since your on Win10 just perform a Fix through the Netfabb on model. Slices fine then. The mesh definitely needed help.

[image]
.

neophyl, 2022-02-18:
When you load that model into PS2.4.1 (what I have installed currently for testing) it reports that there are 390978 open edges.  Not surprised its having trouble.

If you use the fix through netfab option to repair your model then it slices ok.  Easiest solution would be for you to fix your model.

KoenVanduffel, 2022-02-18:
@neophyl Thankyou. I am clearly not a PrusaSlicer adept as i missed the warning...
Surprised Fusion 360 produces models that are not manifold. Cura does not report any issue and just slices the model "correctly" PrusaSlicer and SuperSlicer are not so forgiving and report the errors. I repaired the model in PrusaSlicer, saved the STL and after reopening have no errors any more. Also when opening the .3mf in Fusion 360, repairing it in Fusion and re-exporting as .3mf or .stl there are no errors. Weird that Fusion produces errors (which it can cure by itself upon re-import).

Thanks for the heads-up learned that i have to check my models in both main slicers.

closed 2022-02-18 by KoenVanduffel