prusa3d/PrusaSlicer #7936: No fill fill makes weird bridges

MikeSavad, 2022-02-18:
So i'm experimenting with bridges for PETG (still no luck). And I noticed that when I don't add any bridge, it does a weird thing, it creates a squiggle over nothing, and connects the bridge to that.  If I put just 1% of fill, it doesn't do that.  Not sure if its a gyroid thing or what. I also noticed that when I choose to see retractions, it puts it in a weird place on the floating side. 

On that note, is there a way for it not to put a retraction wipe over the bridge? I have a little dot where it seems to be doing something and I think it might be messing up my bridges. Seems other programs do that too, but there must be a way to prevent it from wiping over a bridge. 

Anyway I think I included all the right info in the packet.

Actually I might have done it twice, I don't see a listing of what I sent.

----Mike Savad

[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/8093165/bug.zip)

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your pri
[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/8093157/bug.zip)
nter, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

MikeSavad, 2022-02-18:
I'm using 2.40 the 3mf should be in the bug packet.

I resaved it. If you slice it, and then go to the first bridge layer. The second from the bottom, I removed all the fill, you'll see a little squiggle, where as the others have bridging all the way across. 

[bridge-Multiple-10ms-flow-1-to-6.zip](https://github.com/prusa3d/PrusaSlicer/files/8097048/bridge-Multiple-10ms-flow-1-to-6.zip)

fe60, 2022-02-19:
@MikeSavad The .3mf file has been there already and the issue was reproducible with your initial post. Please excuse my unnecessary request.

The "squiggle over nothing" disappears as soon as the "Ensure vertical shell thickness" option has been deactivated, but the sliced result still looks strange. The screenshots below show the:

- the infill right before the bridge
- the bridge with and
- the bridge without "Ensure vertical shell thickness".

[image]  [image]  [image]
0% infill in the back and 10% infill in the front

MikeSavad, 2022-02-19:
I found just by adding 1% fill will make it go away, even though there isn't any visible fill there. I haven't tried printing it to see what it does. My guess is, at that size I wouldn't see anything much, but it would probably add to the blobbiness. 

The wipe retraction over the bridge is also bothersome and may be creating issues for me.  Also weirdly when I make these, (and I only saw it once because I wasn't paying attention the other times), by the 4th bridge, towards the end of printing it, the fans turned themselves off for some reason.  Like it had 2 more passes and I saw the fans turn off. 

I don't know if that's related or what that's about, but all the samples I made had the same characteristic, that the first 3/4's laid out flat (for a bridge anyway), and the rest came out messy, or it looked like it wasn't cooled enough. I'm not sure what that was about. 

What is ensure vertical thickness?

----Mike Savad

fe60, 2022-02-21:
For the **no infill** part of your issue there is at least a workaround.

> adding 1% fill will make it go away

The documentation already says, there are known issues with "Ensure vertical shell thickness" (see documentation snippets below). Let's wait for feedback from someone knowing more details.

<details>
<summary>Help content about "Ensure vertical shell thickness"</summary>

Tool tip in PrusaSlicer:
> Add solid infill near sloping surfaces to guarantee the vertical shell thickness (top +  bottom solid layers).

[Prusa Knowledge Base](https://help.prusa3d.com/en/article/layers-and-perimeters_1748#ensure-vertical-shell-thickness):
> This function solves one of the biggest issues of older slicers, and that is having holes between the perimeters on a slanted surface. This was typical when printing busts and other organic looking models. Such objects typically had a few holes on top of the head. With this function enabled, PrusaSlicer makes sure it lays down the needed (internal) supports for perimeters in an upcoming layer.
> :warning: This feature is currently still affecting the G-code generation even when off and can produce unexpected G-code. This is a known issue and we are working on a fix.
</details>

For **wiping over bridge** and **fans switched off too early** I recommend to create separate issues (or vote for/subscribe to existing ones). Mixing up multiple problems in one issue can result in confusing discussions.

There are two issues that seem to address the wiping thing already:
Bridge not connected - Don't wipe over bridge #2190
[Bug/Feature request] Disable nozzle wipe on the bridge #7804

MikeSavad, 2022-02-21:
So based on that  i'm guessing the wipe thing is something I have to live with considering it was apr 2019. If I could fix the fan issue that would be good. I have to see if it happens to the current models i'm making today, because I know they will want a model.

Thanks.

----Mike Savad

closed 2026-09-29 by prusaslicer-bot[bot]