prusa3d/PrusaSlicer #7943: Slicer doesn't load defaults after opening a project file.

MikeSavad, 2022-02-19:
If I load up the program using a 3mf, change something. But don't save it. The next time I load, the defaults I had it set to last reverts to something else. It goes to some random default and generic pla's.  I can't seem to get it to stick to what I had it programmed too. 

I wish there was a way to lock that in place so I can't accidentally change something. Because sometimes I've saved over my default while customizing a 3mf file, not realizing I just did that.  Because there is no way to lock in defaults.  Does that make sense?

I can't get rid of some of the defaults. Like I have a blue tree printer, I customized it. I can't get rid of the default bluer printer I got the specs from. It won't let me, so it defaults to that. 

Or in printer it will default to 1.2 thickness instead of my custom one. And this gets annoying fast because I am experimenting on things and I may not remember what my latest one was. I have to now date the save files because it doesn't even show it in the last saved order.

----Mike Savad

TravisWilder, 2022-02-20:
> If I load up the program using a 3mf, change something. But don't save it. The next time I load, the defaults I had it set to last reverts to something else. It goes to some random default and generic pla's. I can't seem to get it to stick to what I had it programmed too.

I have seen this with 2.4.1 beta 1 as well

MikeSavad, 2022-02-20:
It drives me nuts.  I mean I think its good that it loads up its own program separate from the original profile. But then just sort of leaves it a mess when I leave. I'm constantly removing old profiles just so i'm not mixed up by it later.  And since the file bar is rather short and my file names rather long, I can't see the entire thing. So its a guessing game which file I need to load up.

----Mike Savad

closed 2026-09-29 by prusaslicer-bot[bot]