prusa3d/PrusaSlicer #7946: Prusaslicer running out of RAM memory

roberttemesi, 2022-02-19:
### Version
PrusaSlicer-2.4.1-beta1+win64-202202010918

### Operating system type + version
Windows 11 (build 22000), 64-bit edition
Total RAM size [MB]: 68,641MB
OpenGL installation
GL version:   4.6.0 NVIDIA 511.17
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 3080/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+ (doesnt matter, problem with PrusaSlicer)

### Behavior
Prusaslicer crash after importing 5th part. Shows error - out of RAM memory (impossible, computer has 64GB RAM)

#### Project File (.3MF) where problem occurs
[image]

FidelCapo, 2022-02-22:
Can you please attach 3mf file and STL file? Do you have background processing enabled? Thank you

sboehle, 2022-02-22:
I have a similar issue.

When i try to slice the attached stl, prusa slicer is crashing. Version 2.4.0 and also Version 2.4.1 beta.
[Prusa_Enclosure_Bottom_Corners-for-440x500mm-plexiglass-v2-flipped.zip](https://github.com/prusa3d/PrusaSlicer/files/8119577/Prusa_Enclosure_Bottom_Corners-for-440x500mm-plexiglass-v2-flipped.zip)

systeminfo

PrusaSlicer
Version:   2.4.0+win64
Build:     PrusaSlicer-2.4.0+win64-202112211614

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 22557), 64-bit edition
Total RAM size [MB]: 68,631MB
OpenGL installation
GL version:   4.6.14761 Compatibility Profile Context 21.30.23.03 30.0.13023.3018
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon RX 5700 XT
GLSL version: 4.60
<details>
<summary>Installed extensions:</summary>
GL_AMDX_debug_output
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_sample_positions
GL_AMD_gcn_shader
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_half_float_fetch
GL_AMD_gpu_shader_int16
GL_AMD_gpu_shader_int64
GL_AMD_interleaved_elements
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
GL_AMD_occlusion_query_event
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_atomic_counter_ops
GL_AMD_shader_stencil_export
GL_AMD_shader_stencil_value_export
GL_AMD_shader_trace
GL_AMD_shader_trinary_minmax
GL_AMD_sparse_texture
GL_AMD_sparse_texture_pool
GL_AMD_stencil_operation_extended
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_transform_feedback3_lines_triangles
GL_AMD_transform_feedback4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
[... 234 lines cut ...]
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KTX_buffer_region
GL_NV_alpha_to_coverage_dither_control
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

foreachthing, 2022-02-22:
@sboehle your STL sliced just fine (RAM peaked at ~730 MB of 32 GB) on Windows 11 and PS 2.5.0 α 0.

sboehle, 2022-02-23:
Thank you very much for the fast answer.
Today i tried to reproduce the issue and it is gone. Now it is possible to slice it with 2.4.0 and 2.4.1 beta.
I don't know what happened. I'm completely confused.

Yesterday i also rebooted my machine - to be sure nothing at the os level is faulty.

Yesterday i played something around with the stl. When i split it into parts and reorganized the parts on the build plate, slicing was working, also. But not with the original stl. 

Therefore please ignore my post - i'm very sorry for the spam.

foreachthing, 2022-02-23:
No worries! 
Just close the issue and you're home free ;-)

roberttemesi, 2022-02-27:
Today i tried to reproduce the issue and it is gone.  Probably OS issue, not Prusaslicer.

Thank you very much for the fast answer.

closed 2022-02-27 by roberttemesi