prusa3d/PrusaSlicer #7949: Duplicate center perimeter generated for 3 perimeter wall

kaetemi, 2022-02-20:
### Version
Version 2.4.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro

### Behavior
Model with wall thickness 1.26 (3 perimeter wall)
Slicer with the default Ender 3 profile, and 0.24mm layer height DRAFT preset

The center perimeter of the walls will be generated twice, once for the outside and once for the inside wall. The resulting print is a horrible mess of blobs everywhere.

#### Project File (.3MF) where problem occurs
[cdrack v29.zip](https://github.com/prusa3d/PrusaSlicer/files/8103352/cdrack.v29.zip)

kaetemi, 2022-02-20:
(Works fine in SuperSlicer, though!)

foreachthing, 2022-02-20:
> The center perimeter of the walls will be generated twice, once for the outside and once for the inside wall. The resulting print is a horrible mess of blobs everywhere.

How could I reproduce it and where would I have to look for duplicate lines?

kaetemi, 2022-02-20:
> How could I reproduce it and where would I have to look for duplicate lines?

Just slice with the default Ender 3 profile, and 0.24mm layer height DRAFT preset.
ALL the center perimeters are printed over twice.

kaetemi, 2022-02-20:
[image]

See the yellow line? It's putting down the exact same yellow line in the exact same spot a second time.

foreachthing, 2022-02-20:
Jup, it does it with your settings, not with mine, though... that's odd. I am using an Ultimaker 2.
But I can't see anything standing out that could cause the issue.

Jebtrix, 2022-02-21:
Doubling of perimeter kind of like https://github.com/prusa3d/PrusaSlicer/issues/7762

As @bubnikv stated in other issue Thin Wall Detection works in this particular scenario. It's definite a narly bug since its hard to even see in preview with certain objects. The other issue was a single perimeter, this is 3. Wonder if its an odd number of perimeters issue.

NGC869, 2022-02-23:
I described the same faulty slicing logic here:
https://github.com/prusa3d/PrusaSlicer/issues/7973

I found also that the "Thin wall detection" feature does not work very well: it lead to a very slowly (or crashing) slicing, also introducing bad printing strategies.

Thank you.

stepikovo, 2022-07-02:
We released [PrusaSlicer 2.5.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha2) today with the Arachne perimeter generator, it should fix the issue, can you please confirm?

closed 2026-09-29 by prusaslicer-bot[bot]