prusa3d/PrusaSlicer #7951: Raft uses "Raft layers" value for interface layer count

comps, 2022-02-20:
### Version
2.4.0

### Operating system type + version
Linux (Fedora 34)

### 3D printer brand / version + firmware version (if known)
N/A, reproducible using default Prusa profiles.

### Behavior
Raft interface layer count seems to be using the "Raft layers" value instead of the "Top interface layers" value, resulting in raft interface layers equal to actual sparse raft layers.

With "Raft layers" set to 10 and "Top interface layers" to 2, I get:

[image]

Setting "Top interface layers" to 0 changes this, though it still renders the sparse layers as interface:

[image]

I'm presuming this is likely a source code typo - I would expect "Top interface layers" to govern the raft interface layers too, allowing one to specify a high sparse raft with a thin interface (ie. to avoid deformation from a very hot bed).

bubnikv, 2022-11-24:
> I am having the same problem in version 2.5. It seems that the number of
> interface layer is always half of the raft layers regardless of the value
> in 'Top Interface Layers'.

: I am having the same problem in version 2.5. It seems that the number of
interface layer is always half of the raft layers regardless of the value
in 'Top Interface Layers'.

That is how the raft works.

I suppose you have an "issue" because you want to print with a large number
of raft layers. Why are you printing with large number of raft layers?

thillRobot, 2022-11-25:
OK, I appreciate the information. I was experimenting with the raft settings, and I want to understand how they affect the slicing.

Can you explain the meaning on the _Top interface layers_ parameter?

closed 2023-03-06 by lukasmatena