prusa3d/PrusaSlicer #7952: gcode_substitutions will cause errors in prusaslicer_config dump at the end of gcode

TravisWilder, 2022-02-20:
### Version
2.4.1 beta 1

### Operating system type + version
Big Sue 11.6.1

### 3D printer brand / version + firmware version (if known)
Voron 

### Behavior
At the end of the gcode in the prusaslicer_config dump the gcode_substitutions is added, but with line breaks
This resulst that the printer will try to execute the M204 which will not work.
The gcode_substitutions itself is working 

`; gcode_label_objects = 0
; gcode_resolution = 0.0125
; gcode_substitutions = "((;TYPE:External perimeter
M204 P2000 T5000)|(;TYPE:Overhang perimeter
M204 P2000 T5000)|(;TYPE:Gap fill
M204 P2000 T5000)|(;TYPE:Thin wall
M204 P2000 T5000))";"$1\\nM204 P2000 T5000";r;;"((;TYPE:Perimeter
M204 P2500 T5000)|(;TYPE:Top solid infill
M204 P2500 T5000))";"$1\\nM204 P2500 T5000";r;;"((;TYPE:Other
M204 P4000 T5000)|(;TYPE:Custom
M204 P4000 T5000)|(;TYPE:Internal bridge infill
M204 P4000 T5000)|(;TYPE:Internal perimeter
M204 P4000 T5000)|(;TYPE:Solid infill
M204 P4000 T5000)|(;TYPE:Bridge infill
M204 P4000 T5000)|(;TYPE:Skirt
M204 P4000 T5000))";"$1\\nM204 P4000 T5000";r;;"((;TYPE:Support material
M204 P6000 T5000)|(;TYPE:Support material
M204 P6000 T5000 interface)|(;TYPE:Internal infill
M204 P6000 T5000))";"$1\\nM204 P6000 T5000";r;
; high_current_on_filament_swap = 0
; host_type = octoprint
; infill_acceleration = 4000
; infill_anchor = 0
; infill_anchor_max = 50`

#### Project File (.3MF) where problem occurs
independet

TravisWilder, 2022-02-20:
From the config.ini it is a one line

```
gcode_resolution = 0.0125
gcode_substitutions = "((;TYPE:External perimeter)|(;TYPE:Overhang perimeter)|(;TYPE:Gap fill)|(;TYPE:Thin wall))";"$1\\nM204 P2000 T5000";r;;"((;TYPE:Perimeter)|(;TYPE:Top solid infill))";"$1\\nM204 P2500 T5000";r;;"((;TYPE:Other)|(;TYPE:Custom)|(;TYPE:Internal bridge infill)|(;TYPE:Internal perimeter)|(;TYPE:Solid infill)|(;TYPE:Bridge infill)|(;TYPE:Skirt))";"$1\\nM204 P4000 T5000";r;;"((;TYPE:Support material)|(;TYPE:Support material interface)|(;TYPE:Internal infill))";"$1\\nM204 P6000 T5000";r;
infill_acceleration = 4000
```

jhlchu, 2022-02-20:
Since you are running a Voron, I'm assuming you're using Klipper. Have you tried using the macro SET_VELOCITY_LIMIT ACCEL= ACCEL_TO_DECEL= SQUARE_CORNER_VELOCITY= instead of M204 macro (if using Klipper)?

TravisWilder, 2022-02-20:
> Since you are running a Voron, I'm assuming you're using Klipper. Have you tried using the macro SET_VELOCITY_LIMIT ACCEL= ACCEL_TO_DECEL= SQUARE_CORNER_VELOCITY= instead of M204 macro (if using Klipper)?

This is not related to the issue as the issues seems to be unrelated to Printer Model also I do not need help on Voron Printer Settings

Jebtrix, 2022-02-21:
Yeah this is from ; gcode_substitutions = section being parsed by your own pattern internally like https://github.com/prusa3d/PrusaSlicer/issues/7955

Using regex start of line (^;Type: .... ) would fix your particular issue in the mean time.

TravisWilder, 2022-02-21:
Thanks this workaround is working fine

TravisWilder, 2022-02-21:
please also consider to exclude the comments as well from substitutions

foreachthing, 2022-02-21:
> please also consider to exclude the comments as well from substitutions

Make this _optional_, please. sometimes I want to do _stuff_ to the comments as well.
Also, optionally, **stop** processing after `; prusaslicer_config = begin`!

bubnikv, 2022-02-21:
@[TravisWilder](https://github.com/TravisWilder)
Would you please provide a 3MF for your original issue so we can reproduce?

TravisWilder, 2022-02-21:
ups I already fixed it by changing by adding the ^ to the regex - and did not saved before
I shown above you see the config.ini part which I used - so you can include this in your config.ini and it should create as well the issue.

bubnikv, 2022-02-21:
Fixed by 

8aefe3fc90687aaa96a0f0fce59d073ad334eb75
Neither the initial comments at the start of the G-code nor the trailing config section will be processed by the G-code post-processor

> Make this optional, please. sometimes I want to do stuff to the comments as well.
> Also, optionally, stop processing after ; prusaslicer_config = begin!

We cannot make everything optional, this would slowly creep to an exponential complexity for both the developers, maintainers, testers, documentation writers and also users.

TravisWilder, 2022-02-21:
great support - looking forward to beta2