prusa3d/PrusaSlicer #7957: max_layer_z not working in toolchange g-code

vcapra1, 2022-02-21:
PrusaSlicer-2.4.0+win64
Windows 11 Pro 64-bit

According to [this page](https://help.prusa3d.com/en/article/list-of-placeholders_205643), `max_layer_z` can be used in custom toolchange g-code.  When I try to use that, however, I get the following error:

[image]

I'm using a wipe tower and before a toolchange I need the head to lift up so it doesn't collide with anything on the way over to change tools.

vcapra1, 2022-02-21:
Oddly, this was added with the others [here](https://github.com/prusa3d/PrusaSlicer/commit/e62bb4a7c7bc93c62ffb05a848fd63f862b5b633#diff-cb3d09d2073b6e9032c65d07aa8f10d166324a53136b2aa1288f7b76f7d94319R371), but only as a comment.

vcapra1, 2022-02-21:
This seems to fix the issue without causing any others- https://github.com/prusa3d/PrusaSlicer/compare/master...vcapra1:fix-max_layer_z?diff=unified.

lukasmatena, 2022-02-21:
Thanks, I'll look into it, I already have a branch with some placeholder parser improvements. We'll fix it for 2.5.0. I will also update the documentation so it reflects current state.

vcapra1, 2022-10-01:
Was this fixed in 2.5.0?  I have been using a custom build of 2.4.0 that fixed the issue, but since 2.5.0 came out I installed that version, and am back to the same error.

vcapra1, 2022-10-01:
Nevermind, I see it wasn't added in 2.5.0.  It appears to be working in 2.6.0-alpha0

closed 2022-10-01 by vcapra1