prusa3d/PrusaSlicer #7958: PrusaSlicer Mac BigSur not possible to change extruder

Nipper37, 2022-02-21:
### Version
_Version of PrusaSlicer used goes here_2.4.0 and 2.4.1 (beta)

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_macOS Big Sur V11.0
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_Original PRUSA i3 MK3S+ +MMU2S

### Behavior
* _Describe the problem_Although correct printer + MMU are loaded it is not possible in the object list (advanced nor expert mode) to change or select the extruder (Pull down menu does not appear ). Same problem right clicking object on the workplace; the menu does not appear, therefore not possible to select an extruder.
* _Steps needed to reproduce the problem_N/A
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_yes

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

https://user-images.githubusercontent.com/62954866/154927697-2b7ed71c-46cc-404f-a7aa-6a81eccfeb7e.mov

FidelCapo, 2022-02-22:
I think you're left-clicking. You need to right click to open the extruder menu.

Nipper37, 2022-02-22:
No I was right clicking. But anyhow solved now. Apple Mouse preferences grrr.

closed 2022-02-22 by Nipper37