prusa3d/PrusaSlicer #7959: [MSVC][std:c++latest][permissive-] PrusaSlicer failed to build on MSVC

QuellaZhang, 2022-02-21:
Hi All,

The MSVC team recently added PrusaSlicer as part of RWC testing. And we found some errors when building PrusaSlicer with '/permissive-' or '/std:c++latest'. Could you please help look at this issue or provide some workarounds? Thanks in advance.

**Repro steps:**

1. open VS2022 x64 Tools command
2. git clone https://github.com/prusa3d/PrusaSlicer.git F:\gitP\prusa3d\PrusaSlicer
3. set _CL_= /std:c++latest /D_HAS_STD_BYTE=0 /D_HAS_DEPRECATED_SHARED_PTR_UNIQUE=1 /D_SILENCE_CXX17_SHARED_PTR_UNIQUE_DEPRECATION_WARNING or set _CL_=/permissive-
4. cd F:\gitP\prusa3d\PrusaSlicer\deps\PrusaSlicer-deps
5. cmake -G "Visual Studio 16 2019" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.18362.0 -DCMAKE_BUILD_TYPE=Release ..
6. msbuild /m /p:Platform=x64 /p:Configuration=Release ALL_BUILD.vcxproj /t:Rebuild (will fail with /std:c++latest)
7. mkdir F:\gitP\prusa3d\PrusaSlicer\build_amd64
8. cd F:\gitP\prusa3d\PrusaSlicer\build_amd64
9. cmake -G "Visual Studio 16 2019" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.18362.0 -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH=F:\gitP\prusa3d\PrusaSlicer\deps\PrusaSlicer-deps\destdir\usr\local ..
10. msbuild /m /p:Platform=x64 /p:Configuration=Release PrusaSlicer.sln /t:Rebuild

**Build log:**
/permissive-: [build (1).log](https://github.com/prusa3d/PrusaSlicer/files/8108402/build.1.log)
/std:c++latest: [build (2).log](https://github.com/prusa3d/PrusaSlicer/files/8108422/build.2.log)

**Error info:**
/permissive-:
F:\gitP\prusa3d\PrusaSlicer\src\slic3r\GUI\Notebook.hpp(248,35): error C4596: 'OnNavigationKey': illegal qualified name in member declaration
F:\gitP\prusa3d\PrusaSlicer\src\slic3r\GUI\Notebook.hpp(248,35): error C4596: 'OnNavigationKey': illegal qualified name in member declaration 
F:\gitP\prusa3d\PrusaSlicer\src\slic3r\GUI\Notebook.hpp(248,35): error C4596: 'OnNavigationKey': illegal qualified name in member declaration 

/std:c++latest:
F:\gitP\prusa3d\PrusaSlicer\deps\PrusaSlicer-deps\dep_wxWidgets-prefix\src\dep_wxWidgets\src\propgrid\props.cpp(2205,54): error C2445: result type of conditional expression is ambiguous: types 'const char [20]' and 'wxString' can be converted to multiple common types
F:\gitP\prusa3d\PrusaSlicer\deps\PrusaSlicer-deps\dep_wxWidgets-prefix\src\dep_wxWidgets\src\propgrid\props.cpp(2201,21): error C2664: 'wxFileDialog::wxFileDialog(wxWindow *,const wxString &,const wxString &,const wxString &,const wxString &,long,const wxPoint &,const wxSize &,const wxString &)': cannot convert argument 5 from 'long' to 'const wxString &'

hejllukas, 2022-02-21:
Hi,

It's cool that PrusaSlicer was added as part of the RWC testing.
Speaking of that, we recently encountered a weird bug in the 32-bit MSVC compiler 19.29.30037 (which is also in 19.29.30140, the current Visual Studio 2019), where the following line https://github.com/prusa3d/PrusaSlicer/commit/878dce1fdf5f52ca8957a1b7fd1f29cb5035f233#diff-acef1dc443e5765865060c01eae58c1c74200e3a59cf644119d830ddead60473L442 always caused the MSVC compiler to crash with the following error:
```
C:\PrusaSlicer\src\eigen\eigen\src\Core\AssignEvaluator.h(631): fatal error C1001: Internal compiler error.
(compiler file 'd:\a01\_work\3\src\vctools\Compiler\Utc\src\p2\main.c', line 213)
```
This crash only happened when we were compiling a Win32 build. It looks to me like some issue with optimization, so we make a workaround for Win32.
Unfortunately, we didn't get around to create a minimal test case for reporting the issue using the proper way. But I thought it was a good idea to at least mention it now.

QuellaZhang, 2022-02-22:
For the ICE, I have reproduced it and opened a bug to track the issue.

QuellaZhang, 2022-06-06:
@hejllukas ICE has been fixed internally, the next or next next release may include this fix (not 100% sure) :)

hejllukas, 2022-06-08:
Thank you so much for creating the bug report for the ICE issue and letting us know that the ICE has been fixed.

The compilation with **/permissive-** was resolved with 33fe1c44b47afa08ae5b1d426be65f944e9aab32 some time ago.

The compilation with **/std:c++latest** hasn't been resolved yet. This issue affects wxWidgets 3.1.4, which we are using. But we are in the process of upgrading to a newer version of wxWidgets, where this issue is already fixed.

lukasmatena, 2024-02-19:
I assume that this can be closed now.

closed 2024-02-19 by lukasmatena