prusa3d/PrusaSlicer #7963: [Enhancement Request] Adjust wipe tower to wipe soluble support material fully before moving to perimeters.

ggppjj, 2022-02-21:
### Version
2.4.1b1

### Operating system type + version
Windows 11 Pro Insider

### 3D printer brand / version + firmware version (if known)
MK3S+/MMU2S

### Behavior
The slicer currently does not restrict perimeter layers on the wipe tower to not be printed with soluble materials. In practice, a print with soluble supports or soluble support interface turned on should avoid printing the perimeter of the wipe tower when the support/interface material is being switched to/from.

To be clear, the slicer doesn't currently appear to print the wipe tower perimeters in the soluble support material, but it doesn't fully wipe the remaining material out of the nozzle before transitioning to the perimeters. In my case, this leads to giant blobs of soluble support all in the back right corner of the wipe tower's perimeter, where the transition from soluble support to PLA happens due to the current wipe behavior.

_Is this a new feature request?_
Yes, in how the slicer deals with soluble materials.

rtyr, 2022-02-22:
> The slicer currently does not restrict perimeter layers on the wipe tower to not be printed with soluble materials. In practice, a print with soluble supports or soluble support interface turned on [...]

PrusaSlicer doesn't print perimeters of the wipe tower with soluble filament since 2.4.0-alpha1.

> To be clear, the slicer doesn't currently appear to print the wipe tower perimeters in the soluble support material, but it doesn't fully wipe the remaining material.

You have to increase purging volumes. PLA/PVA usually doesn't work with default purge volumes. 

> In my case, this leads to giant blobs of soluble support all in the back right corner of the wipe tower's perimeter, where the transition from soluble support to PLA happens due to the current wipe behavior.

Giant blobs? You have probably incorrectly calibrated IR sensor.

It mostly seems like misconfigured purge volumes and incorrectly calibrated IR sensor. Please note, that some PVA materials are very soft, which will greatly affect IR sensor sensitivity.

Please contact our tech support via live chat at www.prusa3d.com for more troubleshooting about IR sensor calibration or purge volume settings.

Also please provide your 3MF project file and some photos of those blobs, so we can check.

ggppjj, 2022-02-22:
Thank you for the response! 

> You have to increase purging volumes. PLA/PVA usually doesn't work with default purge volumes.

I'll have to do some additional adjustments to my PVA temps as well I think, but I'll be sure to keep this in mind also. Hopefully some more refinements to the purging volume setup are in the pipeline, it's all too easy to forget to set this up per-project sometimes. 

Here's what they look like: 
[image]

The behavior that I'm seeing that I think is contributing to the blobs can be seen in the attached gcode (3mf also included) at (for a single example) layer 234. It starts off printing the soluble support layer from having already been loaded from the layer before, moves to the wipe tower, unloads, loads PLA, and seemingly interrupts the remaining purge to print the perimeter. It seems that the interruption to the purge is happening before the nozzle is fully purged of PVA, leading the remnants in the nozzle to become part of the wipe towers shell until it finishes transitioning back to PLA.

I'm trying to upload the 3mf and gcode in a zip file, but GitHub won't allow me to upload it.  
[image]

ggppjj, 2022-02-22:
I've uploaded the zip with the gcode and 3mf to Google Drive, please let me know if this is acceptable. 

https://drive.google.com/file/d/1Qy_RPE_qV7uc_WXK5Engvg4gqGn1-mB2/view?usp=sharing

ggppjj, 2022-02-22:
Apologies, the gcode in that zip is correct, but the 3MF is not. The 3MF in isolation is here:
https://drive.google.com/file/d/1hYU6s6awUaJIvFajETWcQxFbHdbhjJic/view?usp=sharing

closed 2026-09-29 by prusaslicer-bot[bot]