prusa3d/PrusaSlicer #7964: Fan fails while printing bridges PETG

MikeSavad, 2022-02-21:
I'm using version 2.40
[bridge fan fail.zip](https://github.com/prusa3d/PrusaSlicer/files/8111707/bridge.fan.fail.zip)

i've been trying to get my PETG to bridge for a while now, the bridge would look good half way, then it looked like garbage the rest of the way. So I watched it all the way through. The bridge perimeters would print, then it would do the fill, it would do about 6 strokes (on these models), then the fan would shut itself off. It would do a stroke or two. Then it would go on, then off, then on, then off right in the middle. And it was consistent on each bridge. 

Then I tried a 90deg angle, and while the results weren't super great, the fan stayed on.

So I made a new batch, these in this file, 3 100mm bridges, first at 0 deg, the second at 180 (still don't get why that's an option), and the last in 90. I was hoping it would stay on on the 180.

I messed around with my fan on my printer, it doesn't handle low speeds well, not unless I give it a jumpstart. But I don't think its machine based.  I am printing at 10ms, and on the cooling page, I had min print speed set to 15ms - could that be related? The min fan was set to 30%-50%. I'm going to try kicking it to 50%-50% see if that does anything. 

Sometimes the fan would simply stay off for all other layers after the bridge which is less than ideal. I don't remember having issues with PLA, so it must be one of these new settings I was told to add.  And all the layers below it are behaving. Or at least there isn't anything out of the ordinary. 

Thing is I looked at the code on the print run through, and I didn't see any special commands anywhere. 

This print will take 30min to print I  think its 20min till the bridges start. But I think you can do it with just one straight bridge.

----Mike Savad

MikeSavad, 2022-02-21:
Also to note, this happens on all sizes of bridges, but I didn't pay attention if speed made the difference.

----Mike Savad

neophyl, 2022-02-22:
Mike you have Keep Fan always On turned OFF.  If you slice your project and then on the preview a screen you can change what the preview is showing you (dropdown in the bottom left corner).  If you had selected Fan Speed you would have seen that the fan is turned off for the layers above the bridging.  No need to print, its info available in the preview.

Turning Fan always on to ON resultsa in the layers above cooling at fan speed 30%.

If you are testing bridges you might also want to try turning off the thick bridges option on the Print Settings>Layers and Perimeters (or you know, just use the search tool to quickly find the right setting)

As for the angle question, 0 is not actually zero, its AUTO.  So if you want to force it to go that direction then 180 degrees would actually be correct.

rtyr, 2022-02-22:
Is neophyl's answer sufficient for you, can we close the issue? I don't see any problem with fan speed behavior in your project.

MikeSavad, 2022-02-22:
I had it set to min 30- max 50, fan was set to turn off (unchecked the top one)

i've tried no thick, and found it was worse as the guide rails seemed thinner.  

0 is auto, kind of wish it said auto because the rest are in degrees.. But OK.

I still don't get why the fan would turn off in the middle of the bridge or even after. How does that make sense? I always want some fan, because its destroying the bridges with the fan off.   Like why would it turn off and on why bridging it? I'm still not getting that. 

I'm looking at fan speed in the viewer, it shows the bridge as a 100% but it stops in the middle of it. But why would it be zero from that point after? Because my settings at the very least would have the fan running at 30 above that. Its only zero at the base. 

You have to try printing it, the graphic only shows a part of the story. Where half the bridge has the fan, and the other half doesn't. 

In any case, what do I set this on so I know I will have the fan running properly?

Thanks

----Mike Savad

rtyr, 2022-02-22:
> I still don't get why the fan would turn off in the middle of the bridge or even after. How does that make sense? I always want some fan, because its destroying the bridges with the fan off. Like why would it turn off and on why bridging it? I'm still not getting that.

> But why would it be zero from that point after? Because my settings at the very least would have the fan running at 30 above that. Its only zero at the base.

Because your profile is configured that way. Fan will run only for layers with smaller print time than 100 seconds. 

[image]

MikeSavad, 2022-02-22:
I can see that, but I don't get how that makes sense. As its based only on time?  So like  a short bridge comes out because the fan stays on because my time was short enough not to set some instruction off. But a long one the fan shuts off for no reason, all because the print time is longer?   

Like if I did a huge box and the top was being done, the fan would shut off because it was longer than a 100 seconds?  I don't get why that is a good thing.  And considering the min fan speed should be at least 30 - why would it shut off in the middle of the bridge and the there after?

When would this feature be useful?

And if I simply say fan always on - then I shouldn't run into this issue right?  Because its really hard adjusting for bridges when it does that.

----Mike Savad

rtyr, 2022-02-22:
Bridges are using "bridge fan speed", 100% fan speed in your case. Next layer printed on the bridge is not a bridge, therefore bridge fan speed is no longer applied. I don't see anything wrong here.

This is only bridge layer in your project.
[image]

neophyl, 2022-02-22:
Mike because you have Keep Fan Always ON UNTICKED that makes it time based.  When printing larger items out of ABS for example it makes total sense to turn the fan off as ambient cooling is enough and you only want the fan on when needed.  

If you looked at the preview you would see that the actual bridging layers (the lowest ones) are printed with the fan at your 100%.  Thge layers above are NOT bridging layers and as the layer time was above your layer time setting the fan was turned off.
The software is honoring your settings.  Its just that you dont seem to get how it works.

Just turn the Keep Fan Alaways on tickbox enabled, as has been suggested to you several times.  What might not make sense to you in the material you are using does for other materials.  If you look at the cooling settings,   That keeps it between the min and max values except where over riden by other settings.

MikeSavad, 2022-02-22:
But how is that an advantage?

Right now it is set on time, I understand that now... But like why is it an advantage for any material for the fan to shut  off randomly while making a long bridge? And then why would it turn on, only to turn off again 7 seconds later? Or on for 5, then off again and then say off when it gets to the next model.

I took benchy and made it huge. With the fan unchecked, it made one layer as no fan at all, all because of how long that layer would take to make. I don't see the advantage. It would obliterate that layer. 

And because there is no warning telling me that my fan will be off, it ended up making a mess. 

With the model above, right after the bridge was made, the fan was turned off.  Not even set to 30%, why is that an advantage to anyone?  I think that's what i'm now wondering.  I mean if I was working with nylon, and I wanted certain areas to not have the fan, then it would make more sense to be able to paint those areas (like I can with the paint can, supports etc), to turn the fan off there. But there are times when you probably want it on.  Like for thin things or detail etc. And all because it took longer than a 100 seconds to make it turns the fan off?

----Mike Savad

MikeSavad, 2022-02-22:
So lets say I need to make a large open span, a bridge, 100mm long but like 200 mm long. Can't have supports for whatever reason. 

I have the fan always on turned off. 

The time it would take would probably be way over 100sec,  but I don't want the fan always on because the material wouldn't like that normally but for the bridge its needed.

Shouldn't there be something in the program that would force the fan to stay on at a 100% while it goes over the entire span of the bridge? And a safety to have some fan as its doing an infill over it so it doesn't obliterate the bridge below?  That would make sense to me. 

----Mike Savad

neophyl, 2022-02-22:
The point being is that except for certain materials you should always have Fan Always on = ACTIVE.  This whole thing is because you have a setting unticked.  Just turn the bloody option on.  Unless printing something like ABS or nylon etc.

Why you turned it OFF in the first place is a mystery.  You also dont seem to be able to understand the decouple between what are bridge layers and something you are calling a bridge.  Bridge Layers are printed at bridge fan speed.  All your projects/pictures etc have always shown that.  The layers on top of those are NOT bridged.  Therefore as far as the software is concerned they are like any other part of the print.  You set the rules so its doing what you ask.  Just learn to use the software.

MikeSavad, 2022-02-22:
Its on now.    I'm not sure why its off other than maybe someone said its better to be off. 

Thing is when the thing is checked to off, the bridge, and I know when its doing the bridge because I had it set down way slow - the fan would become intermittent.  Its supposed to be 100 but it didn't act that way when I did the actual print.  So ignoring the top layers, why would the fan turn off in the middle of the bridge? Then on and off again?

That part doesn't make sense. 

----Mike Savad

rtyr, 2022-03-30:
I consider this explained/answered. Closing.

closed 2022-03-30 by rtyr