prusa3d/PrusaSlicer #7965: add  option ironing_extraload_after or similar to permit a good restore printing after ironing in intermediate layers.

iviar, 2022-02-21:
### Version
All versions (I assume until this is fixed/improved) including (verified) 2.4.0

### Operating system type + version
linux (irrelevant detail)

### 3D printer brand / version + firmware version (if known)
(just for your curiosity)
CR-10 (the frame only), modified with: dual Z axis connected by chain loop, own modified Marlin on BTT SKR1.3 controller and TMC2208 drivers (actually) and TMC2209 for Z
(despite requested and provided, none of the above I think is specifically related to the issue)

### Behavior
* when Ironing "all top surfaces", if there are areas that continue immediately after ironing one at an intermediate layer, the column over the first continued area results fragile because during the preceding ironing all filament that could, inside the nozzle melted and partially drained out during ironing, so... when continuing after ironing the nozzle has not enough filament for a little while and the layer adhesion is compromised especially on small next surface.
* There should be an option to extrude/reload a certain amount of filament after ironing (Has also to be found a good default)
* _Steps needed to reproduce the problem_
Print a "J"(serif font)-section shaped object with Ironing enabled on **all top surface** and check layer adhesion resistance, it will likely break at the opposite of the local maximum (ironed)

Expected Results: uniform resistance through any layer, independently if an ironing happen in that layer somewhere nearby
* _Actual Results_
 
[image]

_Is this a new feature request?_
Yes, it's a new feature request

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
[iviar-supportoTV-02-gancio-front.3mf.gz](https://github.com/prusa3d/PrusaSlicer/files/8112225/iviar-supportoTV-02-gancio-front.3mf.gz)
To be precise the one that broke in the image was oriented 90 degree compared to the attached .3mf and had essentially layer height 0.10mm

dartrax, 2022-03-23:
I think what you are seeing is the same as discussed here:
#5904
#4950
#5113
#5634

stepikovo, 2022-03-23:
I'm closing, this is a duplicate of #5634 and #5904
This is most likely caused by a heat creep, for more: https://help.prusa3d.com/en/article/ironing_177488#downsides

closed 2022-03-23 by stepikovo