prusa3d/PrusaSlicer #7968: Extends the top layer

mrGooDMaster, 2022-02-22:
Version 2.4.0
Extends the top layer inside the model. And prints in the air.
[image]
[image]

[3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8116532/3mf.zip)

cfytrok, 2022-07-10:
Same problem. Is there a way to reduce extention of top and bottom layers? Cura works better.

[image]
[image]

mrGooDMaster, 2022-07-10:
> Та же проблема. Есть ли способ уменьшить расширение верхнего и нижнего слоев? Кура работает лучше.
> 
> [image] [image]

Use 3 perimeters, then the problem disappears.

cfytrok, 2022-07-10:
Didn't help.

stepikovo, 2022-07-26:
I'm closing this as a duplicate of #569

closed 2022-07-26 by stepikovo