prusa3d/PrusaSlicer #7971: Monseignor

gryphonB-2, 2022-02-22:
### Version
2.4

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Linux Mint #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
Hi, Hope this is coming through, it appears the stylesheet for github is not loading, making this less than easy.
Anyway, this is about the scaling of prints.
As you're aware, FDM prints suffer from shrinkage in the X & Y. I'm looking for a good way to compensate for this somewhere in the tool chain. One place would be in Prusa Slicer. 
The issue I'm reporting is to do with the scaling feature. It appears the values entered do not persist as I would expect, getting wiped after the slicing. The g-code produced is scaled as expected, but unlike other print parameters the scaling is reset to 100% in the Object Manipulation panel. Personally, I find this behaviour unexpected, inconsistent and not useful and request it be changed to become persistent. 
Since it is a feature request, not a bug report and likely a normal part of Prusa Slicer behaviour there is no point in uploading any files. 

Sidenote, could you inform github about the stylesheet issue? I tried both Firefox & Chromium browsers. Apparently in the new MS github, errors do not occur & therefore reporting is unnecessary.  

* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2022-02-25:
> The issue I'm reporting is to do with the scaling feature. It appears the values entered do not persist as I would expect, getting wiped after the slicing. The g-code produced is scaled as expected, but unlike other print parameters the scaling is reset to 100% in the Object Manipulation panel.

I was not able to reproduce it. Isn't it possible that you used cut tool along the way?

rtyr, 2022-03-30:
No response. Closing.

closed 2022-03-30 by rtyr