prusa3d/PrusaSlicer #7972: Print Fan Error at resume after g-code pause

Griffinator2020, 2022-02-23:
### Version
_Version 2.4.1-beta1+win64_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
PrusaSlicer
Version:   2.4.1-beta1+win64
Build:     PrusaSlicer-2.4.1-beta1+win64-202202010918

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 17,031MB
OpenGL installation
GL version:   4.6.0 NVIDIA 451.48
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1050/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
PRUSA MK3S+ MMU2, prusa3d_fw_3_10_1_MK3S_1_0_6_MMU2S

### Behavior
* On Resume, Err: PRINT FAN ERROR
* Slice model and add M601 to pause print at specified layer, when print is paused (inserted magnet into cavity), click Resume, receive err., try Resume again several times same result. Workaround is to Tune Fan Speed to any value, Resume. Print Resumes.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  * 
[image]

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save`
[Screenshot slicer.docx](https://github.com/prusa3d/PrusaSlicer/files/8121212/Screenshot.slicer.docx)

Griffinator2020, 2022-02-23:
[magnetic recycle label.zip](https://github.com/prusa3d/PrusaSlicer/files/8121574/magnetic.recycle.label.zip)

rtyr, 2022-02-23:
This problem is not PrusaSlicer related. The issue should be reported here https://github.com/prusa3d/Prusa-Firmware/issues, but I believe there are already similar issues https://github.com/prusa3d/Prusa-Firmware/issues/2889, https://github.com/prusa3d/Prusa-Firmware/issues/2764.

Closing here.

closed 2022-02-23 by rtyr