prusa3d/PrusaSlicer #7973: Single walls lead to unwanted double nozzle pass.

NGC869, 2022-02-23:
### Version
2.4.0

### Operating system type + version
Windows 10

### Behavior
A blend of a bug and a feature request.

**The scenario**: printing a thin object, designed with a thickness that match the nozzle size, with a single nozzle pass without "vase mode" or harmful nozzle jumps.
In the attached image you can see a 0.9 mm costant thickness ribbon-like STL: the best way to print this kind of objects is to print each layer with a single nozzle pass, then raising the nozzle to printing the next layer starting from the path end of the previous layer.

**The problems**:
- For the attached example I'm able to obtain the desired result (with a solid, not fractured, extrusion path) setting "1 perimeter" and all the extrusion widths to 0,86 mm, BUT the slicer produce an unwanted double pass for each layer, as you can see from the image!
- I'm aware that one should use the "Detect thin walls" feature, BUT if you activate it the slicer becomes tremendously slow and the nozzle will do some unwanted jump that lead to a bad quality results (for example it start to print from the middle of the STL, while it should start to print from an extremity of the STL.)

Thank you.

P.S. It's seems to me that this inefficient slicing approach affects also the cases where you need to print objects that are 3 perimeters thick, leading to unintelligent printing strategies.

[image]

Riconec, 2022-06-29:
any updates? It drives me crazy because thin wall detection makes single wall but it won't connect to model and prints separately and without it it makes 2 perimeters 90% overlap making a mess in a print. Checked Cura and Simplify3D they can easily make 1 line thin wall and still make a proper print

NGC869, 2022-06-29:
> any updates?

This problem is long been known, and I'm sad that is still unfixed because make PrusaSlicer truly unsuitable for some kind of prints! Anyway the community work is not paid, so we have just to patiently wait for a solution.

NGC869, 2022-06-30:
> We released PrusaSlicer 2.5.0-alpha2 today with the Arachne perimeter generator. Thanks Ultimaker for keeping Cura open and to the Cura development team for their dedication.

Hi bubnikv, thank you so much, **a truly shining good job**!
Using the 2.5.0.-alpha2 I'm able to print the ribbon depicted using a truly single nozzle pass!

Have a good work and let me suggest also to consider this linked behavior: https://github.com/prusa3d/PrusaSlicer/issues/8441

bubnikv, 2022-06-30:
Credit goes to @BagelOrb from the Cura team and to @hejllukas from our team for porting.

closed 2022-07-02 by stepikovo